pinout-xyz/Pinout.xyz #2: Build instructions

Ardakilic, 2015-03-26:
Hello,

I want to build the pages using `generate-html.py`.

I'm running `python generate-html.py` and `python generate-html.py en-GB`, but in both cases I get this error:

```
..after mappings..
Rendering pin pages...
Outputting page pin1_3v3_power
Traceback (most recent call last):
  File "generate-html.py", line 364, in <module>
    with open(os.path.join('output',lang,'pinout','{}.html'.format(pin_url)),'w') as f:
IOError: [Errno 2] No such file or directory: 'output/en-GB/pinout/pin1_3v3_power.html'
````
I'm assuimg I'm missing something.

Markdown module for python is already installed. Using python 2.7.6 on Mac OSX Yosemite if that'd matter.

Do I have to do something beforehand?

Can you please also share some build instructions?

Thanks in advance,

Gadgetoid, 2015-03-26:
Whoops, looks like there's a bug there somewhere! Will look into it.

Gadgetoid, 2015-03-26:
Fixed here: https://github.com/Gadgetoid/Pinout2/commit/bef0b2bbb792b6c1429011b32a86515463b13810

Ardakilic, 2015-03-26:
Thank you, @Gadgetoid for the prompt response, now, all files are built successfully.

Time to translate! :smile:

closed 2015-03-26 by Ardakilic