pinout-xyz/Pinout.xyz #3: Generating HTML error

Ardakilic, 2015-04-01:
Hello,

Firstly thanks for open-sourcing the codes of this project! I have translated all the files to Turkish, but when I try to generate HTML files in my native language I encounter an error:

```shell
python generate-html.py tr-TR                                     [±master]
Traceback (most recent call last):
  File "generate-html.py", line 336, in <module>
    overlays = map(load_overlay,overlays)
  File "generate-html.py", line 89, in load_overlay
    loaded['long_description'] = load_md('description/overlay/{}.md'.format(overlay))
  File "generate-html.py", line 158, in load_md
    html = markdown.markdown(open(filename).read(), extensions=['fenced_code'])
  File "build/bdist.macosx-10.10-intel/egg/markdown/__init__.py", line 494, in markdown
  File "build/bdist.macosx-10.10-intel/egg/markdown/__init__.py", line 359, in convert
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc4 in position 71: ordinal not in range(128). -- Note: Markdown only accepts unicode input!
````

I think there has to be a setting for encoding and decoding.

Here's my fork, along with updated translations:

https://github.com/Ardakilic/Pinout2/

I'd appreciate if you could investigate this.

Thanks!

Gadgetoid, 2015-04-01:
Looks like my generator, or at least the Markdown processor, just can't handle some Turkish characters! I'll see if I can find/add a preprocessor that escapes them before passing them into Markdown. Hopefully not a tricky fix! Also, thank you for your translation efforts!

Ardakilic, 2015-04-01:
Thanks @Gadgetoid , all I could find was adding `# -*- coding: utf-8 -*-` to the header file for UTF-8 support (Simple Python scripts that I've written require this even for comments) and before processing, adding `encode('utf-8')` to the content, but I wonder if either / both should work.

Gadgetoid, 2015-04-01:
Try adding this to the top of generate-html.py, right after `import pinout`

```python

reload(sys)
sys.setdefaultencoding('utf8')

```

Gadgetoid, 2015-04-01:
Seems to work for me, really nice work! The hack that I added to make "Ground" pins all redirect to /pinout/ground is broken though, since that page has a different name/URL. I'm planning to rewrite it to be less hacky- IE: adding a "force page url" into the Pin database perhaps.

Ardakilic, 2015-04-03:
Hello,

Sorry for the late reply,

Indeed, your hack works like a treat, I can generate HTML files flawlessly.

As you mentioned, only thing remains now is the hack for the ground pins that you've mentioned.

Should I make a pull request now ?

Thanks again,

Gadgetoid, 2015-04-03:
Go for it! Also need to figure out how to provide navigation between the different languages! Hopefully I'll find time to fix those things this weekend.

Gadgetoid, 2015-04-10:
I've been working on tweaks, but I've also been making some big structural changes to switch to YAML instead of JSON since it's much less messy- and also to embed the YAML into the Markdown files so everything is kept tidily together.

If all goes well, I'll update this repo, you can submit a pull request and then I'll convert all your JSON files to YAML and get rid of some of the redundant folders.

Ardakilic, 2015-04-11:
Sorry for the late reply. I was busy with my work.

Cool! I'd love to test YAML versions :)

I have created a pull request for the translation. The repository was for the community site www.raspi.gen.tr ( http://pinler.raspi.gen.tr still holds pinout v1 :smile:  ) , so the Turkish layout file holds some extra links. Feel free to delete them.

Thanks,

Gadgetoid, 2015-11-08:
Ahoy. I'll be hosting the Turkish layout on tr.pinout.xyz soon, but I'll be happy for those extra links to stay. They're relevant to the Pi in your region, so why not!

I'd like a bit of your time to translate a few extra overlays and some text snippets soon, if you can spare it, I'll keep you posted!

closed 2015-11-12 by Gadgetoid