pinout-xyz/Pinout.xyz #6: Translation

Maslor, 2015-05-24:
Do you need an english to portuguese (pt-PT) translation? Let me know if you do as I may have the time to work on it.

RogueM, 2015-11-08:
@Maslor yes, thanks for offering your help, I'm sure that would be welcome!

@Gadgetoid is starting to deploy the localised versions of pinout at this stage via pinout.xyz and I'm sure it would help many people in your country if they had the opportunity to read the boards description in their language!

Gadgetoid, 2015-11-08:
@Maslor absolutely! I've finally had time to prepare pinout for a proper translation and update. Every translation will get its own subdomain on pinout.xyz (pt.pinout.xyz) and I'll be working on making sure the translated pages are well linked and indexed so people can find them.

RogueM, 2015-11-14:
@Gadgetoid, you can close this issue I think, for clarity.

closed 2015-11-15 by Maslor