pinout-xyz/Pinout.xyz #9: Question about setup in nginx

BruceJohnJennerLawso, 2015-08-20:
I was hoping to mirror Pinout on my raspberry pi home server running on nginx on a raspberry pi 2. I cloned Pinout2 to the /var/www/html/ directory, installed the python dependencies, then ran the generate-html.py script. When I try to view the pinout.html file generated, I see this,

[image]

Is there some additional dependency that needs to be installed for the site to work normally?

RogueM, 2015-11-08:
having tripped on this myself recently, I'll try a tentative help here:
you need to include the 'resources' folder in the 'en-GB' folder generated by the script.

There is also a MakeFile in this repo that does it all for you.

if problems persist, post again, and I'll poke the wiz himself, @Gadgetoid , for a definitive answer :D

Gadgetoid, 2015-11-08:
I have updated the build scripts tremendously now. You should run `make LANG=en-GB` and you'll get everything you need to host this locally.

BruceJohnJennerLawso, 2015-11-11:
Thanks Gadgetoid, it works now.

99.236.135.151/pinout/output/en-GB/pinout.html

Forgive my sketchy webserver without a real domain name.

One small thing, the html file has files in the resource folder referenced as 

<link href="/resources/print.css?v=1447277472" rel="stylesheet" media="print">

could you change it to something like

<link href="./resources/print.css?v=1447277472" rel="stylesheet" media="print">

which makes more sense than dropping it in the root of the file system?

Gadgetoid, 2015-11-11:
I might be able to add options to the build script which can slipstream in alternate resource URLs for this purpose. You *could* edit https://github.com/Gadgetoid/Pinout2/blob/master/src/en-GB/settings.yaml

Try changing:
```
resource_url: /resources/
```

To:
```
resource_url: ./resources/
```

And rebuild!

closed 2015-12-03 by RogueM