pinout-xyz/Pinout.xyz #11: structure of the YAML files

RogueM, 2015-11-07:
I think it would be good to have a 'version' info in the board YAML file. This would help in a number of ways but I'm thinking for my own needs to be able in the future to be able to assess which revision I might have translated.

so I guess something like 'Revision: <datestamp>' would be useful... can this be added?

lurch, 2015-11-20:
IMHO any hand-edited "Revision" field will be easily forgotten about, and soon become out-of-date (and therefore pretty useless).

If your intention for this is to help keep the translated files 'up to date' with their English counterparts, what about adding a comment into the YAML file with the git commit-hash of the last version of the English file that the file was translated against?
E.g. if you navigate to https://github.com/Gadgetoid/Pinout2/blob/master/src/en/overlay/dots.md and then click on the **History** button, you can see that the last (abbreviated) commit-hash of the file is `d69e9ff`. So if you decide that https://github.com/Gadgetoid/Pinout2/blob/master/src/fr/overlay/dots.md is up-to-date compared to the English version, maybe you could add a comment to https://github.com/Gadgetoid/Pinout2/blob/master/src/fr/overlay/dots.md saying something like "Last translation up to date with d69e9ff" ?  \*shrug\*

Gadgetoid, 2016-10-19:
I'm guessing for the time being we wont do anything with this? Safe to close this issue for the time being?

The whole site, static site generator and way things are translated could use some serious TLC anyway.

RogueM, 2016-10-22:
yes... I think it would be non-effective in practice. I don't really have a solution, it's up to translators, I guess, to spot outdated content. Most content revision involve the tags anyhow and their additions typically break the non-en lang if absent, so it's usually a case of fixing them when we had features (it's a PITA, but the way it is).

Any other type of content discrepancy are typically minor annoyances, though tere have been a very small number of cases where tweaks to the en overlay was to address pretty serious errors of a technical nature (Red Bear IoT pHAT and sense HAT come to mind).

closed 2016-10-22 by RogueM