pinout-xyz/Pinout.xyz #13: Generating HTML procedure

RogueM, 2015-11-07:
when time permits, could you write up the procedure for generating HTML for local check of the Overlay (I assume you can make a local copy to preview translation work most notably)?

I guess a md is needed for less than apt people like myself :D

RogueM, 2015-11-07:
OK, maybe I'm not as stupid as I look but I'm stuck nonetheless!

python ./generate-json.py fr-FR
python ./generate-html.py fr-FR

I get what seems to be what I want in the 'output' folder, but it's a bare version with no styling. Web dev is definitively not something I know anything about so I'll need hand holding from there on (assuming the HTML I generated are sound using the above steps).

Gadgetoid, 2015-11-07:
I set up the Makefile in order to run builds, it runs the relevant build steps ( well there's only really one step at the moment ) and copies the resources into the output folder. To do it manually you can just:

```
python ./generate-html.py fr-FR
cp -r resources output/fr-FR/
```

This is useful if you don't have Make installed.

I've just updated the Makefile to include a `LANG` parameter, so if you have make, you can run:

```
make LANG=fr-FR
```

RogueM, 2015-11-07:
doh! Totally missed the Makefile *facepalm*

RogueM, 2015-11-07:
I've just tried your tweaked MakeFile and it works fine, but the HTML relationship in the 'output' folder does not seem right...

my base pinout.html is at file:///Users/rogue/Desktop/Pinout2/output/fr-FR/pinout.html
ay link I click try to access file:///pinout, for example file:///pinout/spi
doesn't seem to find the stylesheet either.

... I feel like such a n00b, should it work or is a Mac/Safari thing?

Gadgetoid, 2015-11-07:
Oh it's more an "it's really weird because I've got it set up in a crazy way on my webserver" issue. Run `make LANG=fr-FR serve" and hit up http://127.0.0.1:5000 to see if that works.

I really need to move to a proper location, perhaps with a domain, and fix this weirdness because it makes a lot of things harder than they should be.

RogueM, 2015-11-07:
yeah! that works...

Gadgetoid, 2015-11-07:
Woohoo! This is what I get for leaving all the documentation in a dusty corner of my brain ;)

closed 2015-11-07 by RogueM