pinout-xyz/Pinout.xyz #14: full translation of overlays

RogueM, 2015-11-08:
I see that that some tags are interpreted with no control over the translation within the overlay files. For example:

Made by Ryanteck
Classic 26-pin
Uses 4 GPIO pins
More Information
Buy Now

... am I missing something or is that on your TODO?

Gadgetoid, 2015-11-08:
I think my plan was to include snippets to handle these, saved in the settings.yaml of each language. It should be relatively easy to implement.

RogueM, 2015-11-08:
cool. also, for the 'pin' files, 'Physical pin x', not that much of a deal, but good if that's possible too!

RogueM, 2015-11-08:
and yep, in the settings.yam makes sense, since that is already where the pin definition sits, so anything global (save for temple html) in the same file would make life simple!

Gadgetoid, 2015-11-08:
Updated in 84c63841c7ff07ee8699036e541daaa4c3cbecbb

See https://github.com/Gadgetoid/Pinout2/blob/master/src/en-GB/settings.yaml for details.

Just copypasta the `strings` section into your French yaml and translate away!

RogueM, 2015-11-09:
some of them are localised now, but you missed one:
'* uses {}  GPIO pins'

and all links type:
'GitHub Repository'
'More Information'
'Buy Now'

... but that's the idea so yeah!

RogueM, 2015-11-09:
you can check my latest commit via pull request but that list should be exhaustive.

RogueM, 2015-11-14:
looks like it's all there now... it's just a question for translators to translate the strings in the settings.yaml (retrofitted for all subdomains as part of PR 48).

closed 2015-11-14 by RogueM