pinout-xyz/Pinout.xyz #15: site localisation issues

RogueM, 2015-11-08:
as noted on Twitter, the UK flag is very difficult to trigger once on the tr-TR
also the flags only appears at http://pi.gadgetoid.com/pinout-tr/pinout

... I think you only just added the functionality, so no rush, just wanted to report it in case you'd not realised. To be honest I'm not sure this is hugely important, presumably, once you've chosen your language you probably don't care to switch again?

Gadgetoid, 2015-11-08:
Hopefully this is fixed in 113dc8ece442fdd5f2c9699573e25a4011f15376

Pages will now link directly to their alternate version, as long as it's available in that language. For example:

* http://fr.pinout.xyz/pinout/piano_hat - links to English and French versions
* http://fr.pinout.xyz/pinout/arduino_spi - links to English, French and Turkish versions

At the moment all languages are live and linked all of the time, but for development purposes I'm planning to add a `status: draft` to the settings.yaml files for pending languages, so they will be ignored by the inter-language URL generator.

RogueM, 2015-11-08:
Works as you describe. Great job!

closed 2015-11-08 by RogueM

lurch, 2015-11-12:
Should http://pi.gadgetoid.com/pinout-tr/pinout now auto-redirect to http://tr.pinout.xyz ?