pinout-xyz/Pinout.xyz #16: potential en-GB data review

RogueM, 2015-11-08:
references to wiringpi that may need review:
/pin-3.md
/pin-5.md
/overlay/spi.md
/overlay/uart.md
/overlay/piborg-ledborg.md

RogueM, 2015-11-08:
the SPI 'pin.md' don't exist. I think it would be a worthy addition to go into some details of the pins, to complement the SPI overlay.

RogueM, 2015-11-08:
/pin-8.md and /pin-10.md have different blurbs, which I guess is not justified. I find the RDX overall more relevant to the average Pi user, but I think a collation of the 2 would be best?

RogueM, 2015-11-08:
pin17 symlink could be made (similar to 4 for 5V)

RogueM, 2015-11-08:
header in pin27.md probably redundant

Gadgetoid, 2015-11-08:
The less redundant opinionated cruft we have the better, really. It means less to translate. The WiringPi stuff was from back when I was pushing it as an alternative to RPi.GPIO but it was too little too late.

I like to keep a decent amount of content on the relevant pages because it's good for discoverability, but it's difficult to figure out what's best.

pin-8.md and pin-10.md can almost certainly be consolidated.

You're right, the SPI pins could certainly do with some elaboration- some detail about the difference between MOSI and MISO may be useful.

The system needs better support for consolidating things like 3v power, 5v power and Ground. At the moment I think Ground is a hard-coded fudge.

At some point I want to rewrite the generate-html.py to make it less of a spaghetti mess and more something which could be expanded to additional add-on boards in future.

RogueM, 2015-11-08:
yes, Ground, is problematic for translation I think, I have to keep the name as Ground in the md for it otherwise it trips.

... don't feel every comment points to a necessity to change anything, but I thought I might as well post various thoughts as I process those files for the FR translation ;-)

RogueM, 2015-11-08:
in /overlay/i2c.md, maybe make a note for older PI:
sudo i2cdetect -y 0
?

Gadgetoid, 2015-11-08:
Sheesh. Yeah that's ancient!

I've been refactoring the translation interlinking, so another crazy push coming soon.

Gadgetoid, 2015-11-08:
I've started tweaking the spi.md. Sleep beckons!

RogueM, 2015-11-08:
SPI = subtle pillow invitation?

RogueM, 2015-11-09:
I fixed the pin related stuff discussed above, to save you the hassle. Namely:
- pin-3.md, pin-5.md (removed wiring pi code)
- pin-8.md and pin-10.md were consolidated
- pin27.md header removed

I'll leave you to populate the spa pins as part of your SPI rewrite efforts :-)

lurch, 2015-11-11:
And similarly the "It's easy to get started writing a digital HIGH or LOW to a GPIO pin" on http://pinout.xyz/pinout/pin3_gpio2 seems a bit out of place?

lurch, 2015-11-11:
And given that both http://pinout.xyz/pinout/pin3_gpio2 and http://pinout.xyz/pinout/pin5_gpio3 are both extremely brief, both saying "XXX is one of the i2c pins on the Pi, *learn more about i2c*.", maybe some (more) of the (shared) "UART blurb" on http://pinout.xyz/pinout/pin8_gpio14 and http://pinout.xyz/pinout/pin10_gpio15 should be moved to http://pinout.xyz/pinout/uart, with the UART pins then saying just "XXX is one of the UART pins on the Pi, *learn more about UART*."?

And I guess the SPI pins (19, 21, 23, 24, 26) could each do with a "*learn more about SPI*" link? (apologies if the latter is already being worked on).
And I dunno if the "Uses 11 GPIO pins" on http://pinout.xyz/pinout/spi makes sense?

RogueM, 2015-11-11:
I think the number of pins is not particularly accuate at the moment, due to some lefovers here and there. I think @Gadgetoid is planning a write up for the spi pins, btw.

Incidentally, thank you VERY much for your feedback so far, it's very important that the kinks are ironed out while it's hot :-)

RogueM, 2015-11-11:
ah, I see, the SPI files references both spi0 and spi1, so yeah, maybe not what you'd expect but you'd have to overwrite the pin counting for those particular case. Phil I'm sure is aware of that limitation, and maybe will work some magic in to help out with this one!

lurch, 2015-11-11:
Phil is of course more than welcome to link to https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md ;-)  (but it looks like some of that page may be out of date already)

RogueM, 2016-07-27:
This one is largely outdated... not necessarily addressed in every respect discussed.. Still more productive to create new issues as they require.

closed 2016-07-27 by RogueM