pinout-xyz/Pinout.xyz #25: UART pin colours

lurch, 2015-11-11:
Why are the UART pins given the same colour as the 3v3 power pins? IMHO they should be a different colour.

Gadgetoid, 2015-11-11:
The simple explanation is: http://ethanschoonover.com/solarized

I used the Solarized colour scheme when I first built Pinout, partly because I was using it at the time, and partly because I wanted a well-researched scheme that would be very visually distinct without being obnoxiously bright.

There's an orange which could be used for 3v3 power, but I think I felt it wasn't distinct enough at the time.

Easy enough to change and give it a try!

RogueM, 2015-11-11:
if you change the colour scheme, I think it would be worth considering the choice of colour-blind friendly options. Not that I am an expert and can offer help in the matter, but I guess there has got to be resources that could outline 'best practice' on the web?

lurch, 2015-11-11:
> There's an orange which could be used for 3v3 power, but I think I felt it wasn't distinct enough at the time.

Or maybe you could highlight the UART pins in violet?

Gadgetoid, 2015-11-14:
I'm trialling them in violet on the live site. I think it looks okay! Will probably tweak this into the Git'd CSS soon.