pinout-xyz/Pinout.xyz #27: Missing translation policy / behaviour?

lurch, 2015-11-11:
I see from https://github.com/Gadgetoid/Pinout2/tree/master/src/tr-TR/overlay that there's no Turkish translation (yet) for the sense-hat. This means when I go to http://pinout.xyz/pinout/sense_hat then there's no Turkish flag displayed (which makes sense, from one POV).

However, I also noticed that e.g. http://fr.pinout.xyz/pinout/spi and http://fr.pinout.xyz/pinout/raspberry_pi_dots are identical to the English versions (which makes sense from another POV).

IMHO it would be more consistent to *either* not link to a non-translated page, if that page hasn't been translated yet, *or* to automatically display a copy of the English page so that there are no 'missing pages' on the language subdomains. (Personally, I'd go for the latter option, so that Turkish vistors can see that there **is** a sense-hat page, it just hasn't been translated to their locale yet).

RogueM, 2015-11-11:
the fr-FR was used as a test last weekend to further the multi-lang approach, so many overlays are yet to be translated.

There is a further pull request awaiting and that will 'fix' a few of those, by providing translations, but until a LANG is fully translated you will get these discrepancies I'm afraid.

RogueM, 2015-11-11:
from my point of view, I think the second option is better because it is likely future translations will occur in fragment. It's of course possible to only submit pages with actual translations, but it takes more organisation from the translating party.

RogueM, 2015-11-11:
so, yes, I agree, showing all pages while browsing a subdomain would be the natural 'fix'. It may also encourage contributions in form of translations of those pages (I know for one, I certainly won't mind if someone steps in and translate a new overlay in French I haven't had time to yet tackle!).

lurch, 2015-11-11:
And it would also mean that the non-translated French "translations" could be deleted from github, without any loss in functionality (and only be added back when they're *actually* translated), which IMHO is a much cleaner workflow.

RogueM, 2015-11-11:
well, no, I meant the opposite. We are (currently anyhow) asking people to clone the en-GB src folder as a base to their work... so they are likely to submit the entire thing as a pull request even though it is not fully translated.

... not featuring untranslated overlays would require @Gadgetoid (or myself) to go through those pull requests in some details and delete anything not yet translated. Or put the burden on the contributors, who may not be comfortable enough with Github to do so (I know I'm only starting to get to grip with the finer options).

lurch, 2015-11-11:
I see. Maybe @Gadgetoid could knock up a little script, to see if any the 'translated' files are identical to the English files (in that same forked branch), and remove them from the PR if so?

And perhaps I should have emphasised the **could** (not should) in my earlier comment ;-)

RogueM, 2015-11-11:
that's possible... as long as that does not create issue for those guys.

I think there is a benefit just having them there though, as it would make it easier for a secondary translator to step in. I think we want the barrier of know-how to get involved to be as low as possible, that can only benefit the aim to keep as much of the EN content being available in as many languages as possible.

RogueM, 2015-11-11:
also, the problem with diffing the pull requests with the EN content is that it will drft rapidly, so that wouldn't be reliable in purging not translated overlays.

I guess a new tag could be introduced in the overlay file that indicates which language the content is in, but I'm not sure it is wirth the hassle.

lurch, 2015-11-11:
> also, the problem with diffing the pull requests with the EN content is that it will drift rapidly, so that wouldn't be reliable in purging not translated overlays.

Yeah, that's why I said that the script should compare against the English files in the forked branch that the PR is coming from, rather than the English files in the base repo ;-)

The problem with merging all files by default (including non-translated ones), is that the 'translated branches' (directories) will then be stuck with the 'old English' files (from the time of the original fork). If the 'mainline' English files then get updated to add more info or corrections, then presumably when (another) translator later comes to update the non-translated files, they'll start from the "old English" files already in their local-language directory, rather than starting from the "current English" files in the 'mainline' directory?

Or course this all just emphasises the importance of #11 :-)

RogueM, 2015-11-11:
Yes, translation of outdated EN content is a problem we are in fact facing right now, though I don't suspect it will be a hugely teething problem. I'm not convinced this would be solving the problem though, we can't force or rnsure that the master repo is being synced to the fork because later translations occur.

RogueM, 2015-11-11:
... I do get your point that if a second translator for a LANG comes into the rescue it would ensure they would have to start from the most-up-to-date version. Or maybe they wouldn't realize those need translation or rather how?

I think I would question in any case say a 2nd es-ES translator coming in and not attempt to liaise with the one marked as contributor for the language in this repo, especially if the fork has had recent activity.

That said, I agree the current system is all but completely sound and I'm all for minimizing the retro-fitting efforts I'm currently performing (which incidentally is just due to us revising EN content on the back of the success launching the subdomains, as opposed to beforehand, you live and learn!)

lurch, 2015-11-13:
Been thinking about this a bit more today...

Is it expected that English will always be the "master version", and then other languages are just translations of that? In which case we could modify `generate_html.py` to always treat the files in `src/en-GB/overlays/` as the 'master list' of overlays when generating the output (translated) files, and if a translated overlay is missing, then it'll just use the English 'master file' as input to create the "translated" output file.

Or is it expected that pages may be directly added into e.g. `src/fr-FR/overlays/`without a corresponding version of the page being added to `src/en-GB/overlays/`first? In which case this becomes slightly trickier...

RogueM, 2015-11-13:
I think the likelihood of an overlay being initially submitted in a language other than English, or directly with an EN counterpart is next to nil... but that's not to say it could never happen.

so, yes, I think considering EN as the 'master version' is appropriate for all but extremely uncommon submits.

RogueM, 2015-11-13:
sorry, meant withOUT an EN counterpart..

lurch, 2015-11-19:
And thinking about this a little more, one of the problems of Pinout2 compared to a project that's designed to be translated from the start (please don't take this as a criticism!) is that with everything all being so free-form, and each 'translation' starting out as a duplicate of the entire `en` folder, it's impossible to know which parts of which pages should be translated or not - it's down to each translator's best guess. And common phrases (like "Ground" or "Reserved") have to be translated over and over again on each separate page.
Still, it's better than nothing! :)

Gadgetoid, 2016-03-23:
I'm not quite sure how to approach translation. It's not outside the realm of possibility that Pinout could be completely re-architected - a Pinout 3 if you will - to better support them, but I'm not sure if this would result in more or less cumulative work.

It may be slightly tangental to the topic, but right now the biggest problem is that the English Pinout look and feel has changed dramatically, leaving the foreign language versions in the dust.  The need to translate the entire HTML template is just another one of the many failings of a platform that's had translation kinda shoe-horned in.

In an ideal world:

* All overlays would exist in English as a primary source, and technical data ( which pins they use ) would originate from here
* Other languages would need include only the title and descriptions, and other bits which *need* to be translated
* The HTML would be templated in such a way that translatable strings are separated
* Any untranslated pages would appear as-is on non-English subdomains, but include a "Help translate this page" link.

But... can we move towards this incrementally, without having to steamroll the existing setup, or should this be a considered, ground-up approach in a fresh branch/repo that uses the existing English data files?

lurch, 2016-03-24:
IMHO the best approach would probably be a traditional "separation of content and formatting", i.e. have HTML templates with no "native language" text in them at all (just placeholder fields, using some kind of template markup), and then separate "data files" (e.g. json) with all the text in them, but no markup / formatting. I guess the "data files" could even signify which fields are "translatable", and which fields are "technical" (i.e. don't need to be translated), and then the translated data-files would only need to include the translatable fields.

And then of course you'd be free to update the HTML template at will, and all the translated versions would magically "just work" :)
Having worked on translation software in a previous job, you need to treat English as 'just another translation'. And a neat way of testing that your translation is working, is to have a script which reverses all the translatable strings in the english data-file into a new data-file, and create a "hsilgne" (english backwards) translated output ;-) (makes it really obvious if there's any parts that aren't being translated, that should)

RogueM, 2016-10-22:
I have sort of refined (or worsen depending on your POV) the workflow of translations, and there is now a 'sync' script in the draft folder that does 2 things:

1) create a copy of any en overlays awaiting translations into a 'translate' folder for each existing lang
2) keep that content up-to-date with edits

it has to be ran manually every now and then but it's better than nothing.

anyhow. It does mean that the /lang/overlay folders (should) only contain translated content. the /lang/translate folders are (should be) complementary.

IMO, the content of the 'translate' folder should be made live, so that the issues Andrew pointed out at the origin of this thread are addressed though. So I guess that means that the rendering should go through both 'overlay' and 'translate' folders for each language.

In addition, it would be great to tag those 'translate' md with a note on the live site requesting help with translation.

... possibly easier said than done, but I think that might be the way forward, if imperfect.

RogueM, 2016-10-22:
I had a go at this and butchered some Python to bend it into shape. It needs formatting but the html rendering will now scan both overlay and translate files for match of boards listed in settings.yaml.

I have uploaded a test for the LANG=fr with a single board, the zero-lipo. It works, but I need to figure out how to handle the 'facets' i.e filters so they can be merged in a fully translated selection.

I guess the way I envisage it would work is that the groups (type) would be defined as translatable strings defined once in the settings.yaml file, as per other translatable strings. But, yeah, I'm not quite and that job might be best left to someone who understand a smidge more about javascript and python than myself (@Gadgetoid?)

RogueM, 2016-10-23:
I ended up getting around the facets/filters issue by treating all overlays in the 'translate' dir as a generic 'misc' type.

I have now implemented this for all lang so ALL boards will appear on ALL subdomains, with those in English featuring a call for help in translating them.

... that means that the flags in the bottom of each page should be consistent across the site, though there would be in effect no point for users jumping back to the en parent since all content from the master en will now be available using a self-contained navigation system to jump from board to board.

lurch, 2016-10-23:
> I have now implemented this for all lang so ALL boards will appear on ALL subdomains, with those in English featuring a call for help translating them.

Excellent, I was hoping you'd add something like that :+1: 

I'm not entirely sure what you mean by the facets/filters thing, but rather than lumping all untranslated boards into the 'misc' category, a more robust solution would probably be to have an 'internal category' which is never translated (and is used to determine which category a board is in) and a 'display category' which *is* translated, and displayed to the user on the website.

RogueM, 2016-10-23:
Basically, all md loaded from the translate folder are filtered on the boards page in the 'other' category that exist in any case to catch board otherwise difficult to classify. That category uses a translatable string, so it won't be an entry in english that stands out like a sore thumb.

It also means that all untranslated md will appear under that `group-other` translatable string on the left, which may or may not be a good thing.

RogueM, 2016-10-23:
Incidentally I did try and sort of succeeded in having translatable strings for all boards type, which I guess is what you suggest, sort of. However, there are several hurdles:

1) this system becomes hard to implement due to type supporting multiple values
2) it requires adjustment to the lang translated strings each type a new category is added

so, yeah, I backtracked on that, preferring a solution that requires the type field to be translated for each board. It has pros and cons, for sure.

For now what I've strung together is merely a system that allows resolution of the bug/issue discussed herein, not necessarily a final, comprehensive solution.

lurch, 2016-10-23:
I guess an alternative solution would be to always use the files in the 'en' folder to determine what category a board is in, and never 'look' at the translated category strings, other than for display purposes?

RogueM, 2016-10-23:
yes, well, that's sort of the same idea though... in the context of a lack of separation between translatable content and technical data?

You go back to a situation where all lang need to have translatable strings though. Right now we don't even have translations for those groups in most languages (in fact, only fr does, simply because I speak French, the latest es translation where submitted without type fields translation).

Gadgetoid, 2016-11-01:
> IMHO the best approach would probably be a traditional "separation of content and formatting", i.e. have HTML templates with no "native language" text in them at all (just placeholder fields, using [...]

I've recently pushed up some changes that go a long way toward making this the case.

`layout.html` and `boards.html` are history, now replaced by `common/layout.html`, `common/page.html` and `common/boards.html`, and you will see some new nomenclature for including "strings" (translated text snippets) and the values of settings in these templates.

All that remains in the templates folder of each language is `footer.html` which is mostly translated text strings and would be painful to separate any further.

If you want to give me a wishlist for translation functionality which I can work through and implement, stick a note against the project here: https://github.com/Gadgetoid/Pinout.xyz/projects/1

I'd really like to tidy up `generate-html.py` to help make sense of this stuff, but oooooo boy is it a tangled web of insanity.

lurch, 2016-11-02:
> stick a note against the project here: https://github.com/Gadgetoid/Pinout.xyz/projects/1

I don't have permission to edit / update / add to that.

What you've written in your previous comment sounds good, but I won't have time to take a look at the code-changes for a few weeks at least :-/

RogueM, 2016-11-03:
I suspect the post was addressed to me... we'll keep this topic open as the place to discuss issues with the translations in any case.

I think the latest changes go in the right direction in term of separating translatable strings from the site 'building blocks', there are just a few more strings to extract out such as the filters label, but apart from that it is pretty much there.

The content, i.e the overlays need more work however, to minimise the possibilities of changes to the master `en` not spreading to the translations. Ideally, anything that is not relevant to translation should be held in the master 'en' solely, and sourced from there when generating the static pages.

I guess, one way to achieve that would be to try to load fields from the `lang` overlay first, to allow any to be localised as needed, and if there isn't a match, source it from the `en` markdown file instead. This would allow deletion from the `lang` files of things like 'manufacturer', 'formfactor', 'image', 'pincount', 'eeprom', 'power' and 'ground', as well as all external links.

so a typical `lang` md would end up looking like this (if no links are localised):

name:
description:
and the pin: section (necessary for labels translation unfortunately).

RogueM, 2016-11-03:
I have now completely separated the strings requiring localisation to a separate `localised.yaml` file. All the files that needs translating can also be found in the `template` folder at this stage. This change can be found in https://github.com/Gadgetoid/Pinout.xyz/commit/04f1c5da93ad7eda1b4038a846472d7662902657.

At this point, `settings.yaml` only includes the subdomain specific settings, and the list of markdown files to render. I'm planning to separate the later in the next few days to a standalone file at the root of the /src folder. I think that makes sense since the markdown files to render is now identical for all languages.

lurch, 2016-11-03:
> Ideally, anything that is not relevant to translation should be held in the master `en` solely, and sourced from there when generating the static pages.

Haha, welcome to my line of thinking ;-)

> I guess, one way to achieve that would be to try to load fields from the lang overlay first, to allow any to be localised as needed, and if there isn't a match, source it from the en markdown file instead.

Or, you could load the `en` first as the 'base', and then 'overlay' the lang on top, so any lang-specific strings would overwrite any `en` strings from the base layer. Probably easier than constantly checking "do we already have this string yet?".

lurch, 2016-11-03:
> I think that makes sense since the markdown files to render are now identical for all languages.

Will that also take account of e.g. http://pinout.xyz/pinout/rtk_motor_controller and http://fr.pinout.xyz/pinout/controleur_moteur_rtk and http://es.pinout.xyz/pinout/controlador_de_motor_rtk being named differently, even though they're localised versions of the "same" page?

P.S. I just noticed that http://fr.pinout.xyz/pinout/controleur_moteur_rtk displays `{{title}}` in the title-bar.

RogueM, 2016-11-03:
no, it's purely a division of what data can be found in what file, and generally trying to avoid duplication where unnecessary, where amending one location requires amending all its 'copies'.

granted I have some tools now to help me assess where data might be 'lagging behind' in the translation, this remains a manual process to check for that, and assess whether those discrepancies are intended (99% they are not, but that 1% means it's not clear cut how to implement an 'automatic fix/syncing').

closed 2017-11-29 by Gadgetoid