pinout-xyz/Pinout.xyz #28: PCM_C and PCM_D ?

lurch, 2015-11-11:
[Pin12](http://pinout.xyz/pinout/pin12_gpio18) is labelled as `PCM_C` - it's conceivable that this could be an abbreviation of `PCM_CLK` (in which case why even abbreviate it?), but given the description on the page, it seems `PWM0` might be a more useful label? (which is what e.g. http://raspi.tv/download/RPi.GPIO-Cheat-Sheet.pdf uses)

And [Pin13](http://pinout.xyz/pinout/pin13_gpio27) is labelled as `PCM_D` which seems to make **no** sense at all??

Gadgetoid, 2015-11-11:
There's definitely something amiss here, I probably had a brainfart when I was adding in the alternate pin functions. I obtained my information from here: http://elinux.org/RPi_BCM2835_GPIOs

Looking over it again, it looks like I messed something up.

lurch, 2015-11-11:
> I obtained my information from here: http://elinux.org/RPi_BCM2835_GPIOs

Which is a page I painstakingly wiki-fied and cross-referenced over 3 years ago ;-)

Gadgetoid, 2015-11-12:
Well I'll be damned, I didn't read the history! Rechecking alt functions and documenting them better is definitely high on my todo list, specially as these errors will get duplicated into multiple languages!

lurch, 2015-11-14:
Now fixed by https://github.com/Gadgetoid/Pinout2/commit/bc72b181d0ed36668af26f110a3015c6d1e47750

closed 2015-11-14 by lurch