pinout-xyz/Pinout.xyz #29: "installer" errors?

lurch, 2015-11-11:
The "one-line installer" on http://pinout.xyz/pinout/unicorn_hat and http://pinout.xyz/pinout/display_o_tron_3000 pipes the retrieved file to bash, but the ` | bash` seems to be missing on:
* http://pinout.xyz/pinout/explorer_hat_pro
* http://pinout.xyz/pinout/pibrella
* http://pinout.xyz/pinout/explorer_hat

Gadgetoid, 2015-11-12:
Zoiks! That's surely a bug.

lurch, 2015-11-12:
...and I guess this is another "fix" that won't get automatically translated to other languages? ;-)
(assuming of course that `curl` and `bash` remain named as they are in e.g. Italian)

RogueM, 2015-11-13:
that's undoutedly my doing. I will fix as I am to blame. And no, since nothing is shared between locale then it has to be retro-fitted in all... it's not a major pain, but it definitely isn't automatic.

lurch, 2015-11-13:
Now fixed and published.

closed 2015-11-13 by lurch