pinout-xyz/Pinout.xyz #30: Dependencies

lurch, 2015-11-11:
Maybe a note could be added to the readme about the build dependencies?
On Ubuntu 14.04 I found (by trial&error) that I had to:
```bash
sudo apt-get install libyaml-dev
sudo pip install pyyaml
sudo pip install markdown
```
before `make` would run successfully.

lurch, 2015-11-11:
...and `sudo pip install flask` is needed for `make serve` to work.

RogueM, 2015-11-11:
yes, that is true. Not sure it's somethingthat can stop anyone in tyeir track, but I'm certainly a proponent of stated dependencies, where that does not detract from the main message.

lurch, 2015-11-11:
The main thing that tripped me up, was that when Python reported it couldn't import the yaml module, I was surprised when `sudo pip install yaml` didn't work, so I had to go searching the web to find which package I had to pip-install that supplied the yaml module ;-)

(which is an annoying side-effect of Python module and package names being totally distinct)

Gadgetoid, 2015-11-12:
Package names like "pyyaml" drive me bonkers. I'm all for adding dependencies and other instructions to the documentation. Perhaps these would be better placed in a Documentation subfolder along with instructions on how to build the site to run locally, how to contribute translations, how to contribute boards, etc. Then we can also push it up to ReadTheDocs.org just for giggles.

lurch, 2015-11-12:
> Package names like "pyyaml" drive me bonkers.

@Gadgetoid It gets even more maddening in some situations!  https://github.com/chaosmail/python-fs/issues/3#issuecomment-149940856

With Pinout2 still being a fairly small project, I'm not yet sure if it warrants a separate Documentation subfolder? With this issue I was only really suggestion a one or two line addition to the current Readme.
But by all means feel free to expand this issue into a more general "better docs please" type issue.

Gadgetoid, 2016-11-01:
Looks like this still needs doing. Will stick it against the project.

RogueM, 2016-11-01:
I made a requirement.txt a while back. Isn't that sort of the 'pythonic' way to seed dependencies? I think it is complete anfd functional?

RogueM, 2016-11-01:
of course, it wouldn't hurt to mention it in the readme ;-)

RogueM, 2016-11-02:
I have now documented the use of requirements.txt in the README, as well as trying to outline the process for both new boards submissions and translations (new subdomains or continuous trickle in the `translate` folders).

... not sure I did a great job of it, but at least it is somewhere now, in some semi-understandable form :-)

closed 2016-11-02 by RogueM