pinout-xyz/Pinout.xyz #31: German flag has wrong link

lurch, 2015-11-11:
If I go to http://pinout.xyz/ and click on the German flag, it just takes me to http://pinout.xyz/ when I suspect it *should* take me to http://de.pinout.xyz/ ?

RogueM, 2015-11-11:
yes, German is not fully integrated, Phil had to go to work ;-)
... it will be done tonight I suspect, although the de. subdomain has a lot more translations to follow (similar to fr.)

btw, did you get Phil's request to join the slavk team, if you wanted yo obviously. We are trying to organize it in better ways so as to distill a digest of important info so it is clear what cerain in-between statues may be in some respects.

... don't feel you have to, but you certainly are welcome to, we truly appreciate your dedication!

lurch, 2015-11-11:
> btw, did you get Phil's request to join the slack team, if you wanted to obviously.

I did, thanks, but I've never used (or heard of) Slack before. What advantages does it offer over GitHub?

RogueM, 2015-11-11:
Slack is an IRC based system, so it is possible to separate the type/subjects of discussions, and who is involved. Not as a matter of excluding people from discussing things, but rather minimise the information in Github to the conclusion of ways forward.

So it is a complement rather than an alternative. We decided to open it up to allow translators, and other contributors to have a more casual manner of communicating, and feel more than just 'bug reporters'. It's by no means necessary for you to join, we're happy to read your reports on github in equal manner, but feel in my be limiting to scope of improvements the system could gain that way.

Gadgetoid, 2015-11-11:
Ah. Yes! The German flag being present is an error- I recently rebuilt the English site with some tweaks and it slurped up the German stuff prematurely. I need to expedite my plan to add a "Status: Draft" flag to the main configs.

Gadgetoid, 2015-11-11:
Believe this is fixed now!

closed 2015-11-11 by Gadgetoid