pinout-xyz/Pinout.xyz #34: Drop the country part of the locale specification?

lurch, 2015-11-11:
Given that the Readme itself says "At the moment cultures are not fully supported, so you can't have src/fr-CA ( sorry! ), and there are no plans for this.", and the domains themselves are set up like http://de.pinout.xyz/ and http://fr.pinout.xyz/ ; would it make things 'simpler' if e.g. https://github.com/Gadgetoid/Pinout2/tree/master/src/de-DE was renamed to just `de` and https://github.com/Gadgetoid/Pinout2/tree/master/src/fr-FR was renamed to just `fr` ?

See also: https://en.wikipedia.org/wiki/ISO_639-1 and https://en.wikipedia.org/wiki/IETF_language_tag and all that good fun... ;-)

Gadgetoid, 2015-11-12:
Totally agree. The rationale behind using language-country codes is lost to the mists of time and probably related to my past experience with multi-language websites and .NET ( eep ) region codes.

It'll be painful, but it might as well be done ASAP!

lurch, 2015-11-12:
http://www.w3.org/International/articles/language-tags/ says "**The golden rule when creating language tags is to keep the tag as short as possible.** Avoid region, script or other subtags except where they add useful distinguishing information. For instance, use ja for Japanese and not ja-JP"

Although http://www.codeproject.com/Articles/15175/NET-Internationalization-The-Developer-s-Guide-to suggests you might be missing out on nearly 432 *billion* potential readers, by not having your website translated into Klingon yet ;-)

lurch, 2015-11-13:
Now merged by @Gadgetoid

closed 2015-11-13 by lurch