pinout-xyz/Pinout.xyz #35: Localisation of URLs

lurch, 2015-11-12:
Possibly more of a comment / question than an issue, but I was surprised to just notice that the Turkish version of http://pinout.xyz/pinout/ryanteck_motor_controller_board is *actually* http://tr.pinout.xyz/pinout/ryanteck_motor_kontrol_kart

Wouldn't it make sense for the URLs themselves to be non-translatable, so that the Turkish version of http://pinout.xyz/pinout/ryanteck_motor_controller_board is actually http://tr.pinout.xyz/pinout/ryanteck_motor_controller_board ?

This is obviously something that ought to be fixed (if you consider it worth "fixing") before lots of sites start linking directly to sub-pages on http://pinout.xyz :-)

Gadgetoid, 2015-11-12:
Technically the URL should be translated, but it does set an interesting and potentially problematic precedent. The static site generator takes into account the different URLs and build the language links to work accordingly, but it does make me wonder what happens if, for example, someone made a mistake with translating a URL which was then linked to but later corrected.

Those may be a "we'll cross that bridge when we come to it" level of problem though. Right now I think having the URLs translated is generally a good thing.

lurch, 2015-11-12:
Hmmm, afraid I'm not too sure of the rationale behind URLs being translated, but I wonder if it'd be possible for e.g. http://tr.pinout.xyz/pinout/ryanteck_motor_controller_board (which currently displays **Not found**) to get redirected to http://tr.pinout.xyz/pinout/ryanteck_motor_kontrol_kart ?

If you think that's a crazy suggestion, feel free to close this issue.

(hmmm, a quick internet search suggests translated URLs may help in terms of SEO (which itself is something I've never really looked into))

> it does make me wonder what happens if, for example, someone made a mistake with translating a URL which was then linked to but later corrected.

I guess theoretically you could scan through the github history, see what names the page ever had in the past, and then set up .htaccess rules to redirect them all to the current page-name? ;-)

lurch, 2015-11-16:
I'm still not sure of the best answer to this one, but one small minus-point is that if a Turkish user spots an error on http://tr.pinout.xyz/pinout/ryanteck_motor_kontrol_kart it's not immediately obvious that the file they need to edit to fix t is *actually* https://github.com/Gadgetoid/Pinout2/blob/master/src/tr/overlay/rtk-000-001.md
(although thinking about it, this is actually a more general issue which applies to English too!)

closed 2018-01-25 by Gadgetoid