pinout-xyz/Pinout.xyz #39: Make "info" overlays distinct from "board" overlays

lurch, 2015-11-12:
IMHO there ought to be some kind of distinction between e.g. http://pinout.xyz/pinout/i2c (which is a "general information" page) and http://pinout.xyz/pinout/unicorn_hat (which is a "add-on board specific" page).
And then maybe you could visually display them with a different background colour or something in the pop-up menu?

Feel free to close if you think this isn't important.

lurch, 2015-11-12:
...although I guess this could then tie into https://github.com/Gadgetoid/Pinout2/issues/16#issuecomment-155765266 with "info" pages not displaying the "Uses XX GPIO pins" section.

Gadgetoid, 2016-01-28:
I absolutely agree, and this should be part of a general UX/UI overhaul because the pop-up menu is both absolutely awful, has poor discoverability and not scalable at all.

I wanted to do several things:

* Split overlays into Add-on, Hardware/Device and Pinout Scheme
* Tag overlays in Add-on into different board types, perhaps: Motor Driver, Audio, etc
* Replace the awful drop-down menu with the 3 top-level categories

RogueM, 2016-01-28:
I *think* I have added tags to overlay in preparation for something like this... maybe not for all lang, but there should be a 'type' entry in the EN stuff, I think.

RogueM, 2016-01-28:
yeah, there's type: content, type: board and type: info... should be more or less what you had in mind, though can be split further from there (for board types as you suggest).

Gadgetoid, 2016-11-01:
I think we're now several evolutions removed from this issue, so I'll file it in the round cabinet on the floor ;)

closed 2016-11-01 by Gadgetoid