pinout-xyz/Pinout.xyz #41: arduino-spi.md incomplete

RogueM, 2015-11-12:
I had left this one for a time my brain was functioning. I think it is today :-)

anyhow, I think you are missing something in there, specifically you reference a MakeFile but it is nowhere to be found.

What about just replacing the part of the page starting: 'This basic Makefile should get you started...' by

"You are now ready to compile the sketch. Visit http://pi.gadgetoid.com/article/building-the-pico-piduino for a run-through of the process!"

RogueM, 2015-11-12:
err, wrong link, I meant http://pi.gadgetoid.com/article/programming-your-pico-piduino

Gadgetoid, 2015-11-12:
This also raises the higher question of whether or not this has any business being on Pinout!

RogueM, 2015-11-12:
yes, indeed. My view is that it would be best cut off entirely as it relies on fairly involved articles that are written in English. I tried to locate French version at my end but there was nothing quite covering it (and didn't feel like starting a blog lol).

so, I think it should be deleted entirely in all LANG?

RogueM, 2015-11-12:
Maybe we should wait for the pending PT and DE translations before doing so though, or we may not realise it is back there when the submit!

RogueM, 2015-11-19:
With the DE fully submitted, shall we nuke the Arduino SPI now?

Gadgetoid, 2015-11-19:
Interestingly, that page gets a fair number of views ( about 10/day on average ) so it might be worth considering how we could improve it and set a precedent for more informational pages like this.

But on the flip-side, more pages = more translation and perhaps guides things like this should just live elsewhere.

RogueM, 2015-11-19:
I think it gets the hits based on the 'arduino' keyword, no more no less, I don't think that means people hitting it are in any way interested in programming their Arduino from their pi, unless you have data to the contrary. 

As a matter of interest, how does the traffic compare to the spi page, which also contains an abridge version (-ish, IMO, more suitable)?

from my perspective, if the 'arduino' traffic can be directed to pinout equally well by not getting bogged down with things like that the better. Otherwise I think this opens the door to anything and everything, why not a 'LED' page or going to the extreme 'how to make a pi'? ;-)

... it also relies way too much on external links that are in English, which is another reason for avoiding that kind of things.

lurch, 2015-11-20:
...maybe we could have a separate area of the site under e.g. `/info` where we could have things like `/info/programming_arduino` which other pages on pinout.xyz would be free to link to (if they wanted), but there'd be no obligation for pages under `/info` to get translated, and we could indeed have pages under e.g. `http://fr.pinout.xyz/info/` that don't have a corresponding English version under `http://pinout.xyz/info/` ?
Just a suggestion ;-)   (perhaps it's more hassle than it's worth)

RogueM, 2016-07-09:
This file got the sack at this point, so closing.

closed 2016-07-09 by RogueM