pinout-xyz/Pinout.xyz #52: Unicorn HAT 'singular' issue

RogueM, 2015-11-14:
the Unicorn HAT only uses the PWM pin. This creates an issue for the 'uses X pins' string that then reads grammatically incorrect (in all LANG).

I'm not sure what the answer is, maybe just add the power pin required. This is not a terrible idea, since it uses the 5V, which is not necessarily common?

Gadgetoid, 2015-11-14:
This is probably a better idea than adding a fudge to switch between plural/singular. Also Unicorn HAT technically uses the EEPROM pins too!

RogueM, 2015-11-14:
yes, but that would be true of any HAT right? I think if that's a valid way forward you might as well have an added value, for any one wanting to wire it externally. Then again you could argue that the dot HATs could also benefit from the additional info (but that could be added as text if we really wanted to).

lurch, 2015-11-15:
Yeah, adding power and ground pins to the board-overlay files was something I was thinking of suggesting too - even though e.g. all the GNDs are electrically connected on the Pi-side of the GPIO header, it might also be useful to know which GND pins actually get used on the addon-board-side of the GPIO header? Especially if e.g. messing about with the black hat hack3r or similar.
(with the same argument being made for all HAT boards showing their connections to the ID pins. As I mentioned to @Gadgetoid recently, a board is officially only a HAT if it has an EEPROM, but Pinout2 currently (mistakenly) labels all 40-pin boards as HAT format, and includes no mechanism (yet) for the overlay files to signify if the board is **acutally** a HAT or not)

Although if a board does electrically connect on the board side to e.g. all the ground and both the 3v3 pins, would we want to count them both multiple times in the "uses X pins" count, or just once each? I guess you could argue either way... (with one being a count of the 'physical' pins, and one being a count of the 'logical' pins)

RogueM, 2015-11-15:
I discussed the matter with @Gadgetoid a few weeks back and ground and power pins are consciously excluded, to avoid clutter. I even went to the trouble of cleaning up some me of the overlays for consistency not long ago.

I certainly wouldn't be a fan of reintroducing them, particularly the ground pins, which we would have a hard time documenting, or enforce documentation of, accurately. There are also cases like the Pi-Dac+, where different revisions of the boards had different electrical connections.

So I would rather no data be provided on that front than data that may end up being 'wrong'. I am not particularly happy having the unicorn HAT bumping its pin count to 2 in the manner I suggested to be honest, as I don't think power (or ground) pins should be counted in that tally, but in this instance that is a cure to the grammatical problem.

EEPROM could or maybe should be a bullet point. Not that I would mind if I f they were part in f the pin out, as long as they didn't count towards the tally... they deserve to contribute to it even less so than power/ground in my view as they are not relevant to applications where a Pi is not part of the BOM.

lurch, 2015-11-16:
> There are also cases like the Pi-Dac+, where different revisions of the boards had different electrical connections.

Crikey, that's something I never considered! :-S

> I am not particularly happy having the unicorn HAT bumping its pin count to 2 in the manner I suggested to be honest, as I don't think power (or ground) pins should be counted in that tally

Fair enough.

> but in this instance that is a cure to the grammatical problem.

Maybe we could just use "Uses X pin(s)" instead to fix the singular / plural issue. When it comes to translations, singular/plural is quite a complex issue anyway e.g. https://developer.mozilla.org/en-US/docs/Mozilla/Localization/Localization_and_Plurals or http://stackoverflow.com/questions/1438093/best-practices-for-singular-word-or-plural-words

> EEPROM could or maybe should be a bullet point.

IMHO @Gadgetoid needs to change his "HAT format" code, so that only boards that are **actually** [HATs](https://github.com/raspberrypi/hats#b-hat-requirements) get labelled as so, and then the usage of EEPROM becomes implicit for all HAT boards. (Maybe I should create a separate issue for this HAT-mislabelling?)

> applications where a Pi is not part of the BOM.

Is that really very likely?

RogueM, 2015-11-16:
> Maybe we could just use "Uses X pin(s)" instead to fix the singular / plural issue.

I think that's a pretty ugly way. The Unicorn HAT is likely to be the only board that is going to cause problem.

>> applications where a Pi is not part of the BOM.

> Is that really very likely?

It depends who you ask... I use HATs on my Arduino quite a bit, but I can also agree that it's not Pinout's vocation to cater for those cases were a Pi is not involved.

I think the best may be to declare the EEPROM pins, not just here but all boards... this would solve a bunch of issues all at once? well, I think your point about the "HAT format" code would still stands and the overlap with '40 pins header' could create problem in the future.

lurch, 2015-11-16:
As I tried to point out, different languages pluralise things in different ways (e.g. [Russian](https://developer.mozilla.org/en-US/docs/Mozilla/Localization/Localization_and_Plurals#Plural_rule_.237_.283_forms.29) ), so even if we fix this (for English) by 'not allowing' any board to use less than 2 pins (so that English always uses the same plural form), that doesn't mean that the translation still might not "look ugly" in other languages. But I guess that's a minor issue, compared to the existing "translation issues" Pinout2 already has ;-)

RogueM, 2015-11-16:
yes, depending on what translations might be put forward in the future, it's merely pushing the issue further along, so what? I would personally rather see at @Gadgetoid spend the little time he has on more meaningful improvements.

All I'm saying in this topic is that there is an issue with one board, and that isn't going to be one we run into often, so what is the most economical way to resolve it?

lurch, 2015-11-16:
> so what is the most economical way to resolve it?

The cheeky answer to that would be to close this issue, and pretend it isn't a problem! ;-D

Gadgetoid, 2016-11-01:
Sounds like a plan :D

closed 2016-11-01 by Gadgetoid