pinout-xyz/Pinout.xyz #53: 'use i2c' not displayed

RogueM, 2015-11-15:
if the pin declaration is as follows:

pin:
  3:
    mode: i2c
  5:
    mode: i2c

then the 'use i2c' tag is not displayed. It has to be this way:

pin:
  '3':
    mode: i2c
  '5':
    mode: i2c

I'm going to fix the boards overlays but presumably that is not intended.

lurch, 2015-11-16:
So... this is because pyyaml e.g. loads `5` as a Python integer, loads `bcm14` as a Python string, and also loads `'5'` and `'bcm14'` as strings.

There are 3 ways to fix this issue:
 * Always hand-edit the yaml files to quote the pin-numbers (as @RogueM has done in https://github.com/Gadgetoid/Pinout2/commit/227ef3cea6127968e992e7f80011693c78c37497 ). However this is obviously a bit tedious and not very robust.
 * Always convert the 'pin numbers' in the yaml files to strings (as I've done in https://github.com/Gadgetoid/Pinout2/compare/master...lurch:always_string_pins?expand=1 ). This seems to be in keeping with the spirit of what the code is originally doing.
 * Always convert the 'pin numbers' in the yaml files to integers (as I've done in https://github.com/Gadgetoid/Pinout2/compare/master...lurch:always_integer_pins?expand=1 ). This is my personal preference. (Why store integers as strings, when they're actually numbers!)

I'll let @Gadgetoid choose his personal preference, and then submit the corresponding branch as a PR and delete the other branch.

RogueM, 2015-11-19:
I believe this has been addressed (so can be closed)?

lurch, 2015-11-20:
As I mentioned in my last comment, it's only been "fixed" because you went through and manually quoted all the pin numbers in all the YAML files. (so the problem will re-appear if some adds / edits a YAML file with unquoted pin numbers)

IMHO it'd be better (more robust) to fix it from the code-side, by either always converting the pin numbers to strings (regardless of the YAML), or always converting them to integers. I've created branches doing things both ways, and I'm just waiting for @Gadgetoid to give his verdict on which one he'd like me to submit as a PR...

RogueM, 2015-11-20:
oh, sorry, I thought one or the other had been merged... difficult to keep track of everything at the moment ;-)

RogueM, 2016-10-22:
closing... not ideal but the overlay syntax has enough idiosyncrasies that we can live with one more, i.e enforcing quotation of pin values, even when numerical.

closed 2016-10-22 by RogueM