pinout-xyz/Pinout.xyz #59: DiscoHAT addition

karrika, 2015-11-16:
Hi, I am currently designing a light and sound control board called DiscoHAT (http://discohat.com). I would love to choose the pins in a way that would work well together with HAT boards providing sound in/out functionality.

There is a HAT board called Cirrus Logic Audio Card at Adafruit that sounds like a cool companion board to my DiscoHAT board. But the pins it uses are not known to me. Perhaps there is some "standard" pins for digital audio in/out. But I cannot find the info here.

The only thing I really need for DiscoHAT is UART TX and 8 GPIO pins. But which GPIO pins to choose? So far I have gone with wiringPi 2 to wiringPi 9. Does anyone have good advice? There is still time to pick the right pins for maximum compatibility with other HAT boards.

lurch, 2015-11-16:
http://elinux.org/RPi_BCM2835_GPIOs#GPIO18 suggests that digital PCM audio is available on physical pins 12, 35, 38, 40.

I also suggest you read up on the [HAT specs](https://github.com/raspberrypi/hats#b-hat-requirements). Only boards that use an EEPROM can be called a "HAT board", which also means only one HAT can be connected to the RaspberryPi at a time. So if your board is designed to be used at the same time as another HAT, then presumably your board *doesn't* have an EEPROM, which means you can't call it a HAT.

RogueM, 2015-11-16:
the Schematic for the Cirrus Logic Audio Card that I think you are referring to can be found here:
http://www.element14.com/community/community/raspberry-pi/raspberry-pi-accessories/cirrus_logic_audio_card

You may also want to have a look at the Pi-DAC+ at pinout.xyz

... choosing the GPIO pins for your board really depends on what other board might seem useful to combine with it. I think you'll definitely want to stay away from i2c (which might not be suitable anyhow, due to pull-up), as well as SPI.

I guess that wiringPi 0 to wiringPi 7 would be a better choice therefore, but I haven't given it much thought besides wanting to advise to stay clear of i2c and SPI.

RogueM, 2015-11-16:
well, I suspect you will want to stay clear go wiringPi 1 too, as pointed out by Andrew it's an I2S pin... maybe have a look at the top part of the header, wiringPi 21 to 23?

karrika, 2015-11-16:
Thank you. You are a life saver. I love this Open Source community.

There is a very nice set of pins available for me that still allows my board to work with several audio HATs. My board also has an id eeprom, so it is a HAT, but the eeprom has to be disabled if I stack it with other HATs. I found a nice set of non-conficting pins for DiscoHAT:
wiringPi 2, 3, 4, 5, 6 (for 26 pin connectors). External push buttons.
wiringPi 21, 22, 23 (for 40 pin connectors). External push buttons.
wiringPi 15 UART TX that I need for DMX output.
wiringPi 12, 14 for ws2801 style LED strip control (just an option in case of no conflict with SPI audio)

Once I get the production up and running I hope to add DiscoHAT to the pinout page.

RogueM, 2015-11-16:
wiringPi 6 would conflict with the Display-O-tron, which may or may not be a big problem for some people. Ultimately you can't expect to avoid all pin conflicts but maybe split the buttons 4+4 with wiringPi 26 instead?

karrika, 2015-11-16:
DiscoHAT is built to work with QLC+. The GUI is exported as a web page over WiFi to your mobile phone or tablet. But in case you have a small band you can also control the chases (lights following the music) with 8 foot buttons. So I don't really care about displays. A great audio board is a different story as I may need to export shows (sound clips synced to disco lights). And I also need sound input to let the lights follow the music.

RogueM, 2015-11-16:
Makes sense. The Pi-DAC+ has a pretty cool add-on on BCM25 (wiringPi 6) though, so unless you value the external buttons on the Disco HAT being on the original 26-way header, moving as many towards the top will minimise friction with various add-on boards out there.

karrika, 2015-11-16:
Oh no. The IR sensor is sometimes on wiringPi 0 (example RaspyPlay4), wiringPi 5 (LIRC examples) and now even on wiringPi 6. This is sooo frustrating. I already reserved wiringPi 0 for IR as it is included on RaspyPlay4 HAT. But I will leave wiringPi 6 free and pick wiringPi 26 instead.

Thanks

lurch, 2015-11-16:
I guess the only way to guarantee maximum compatibility with all boards, would be to have some kind of "patch panel" with jumper leads, to allow the user to choose which GPIO pins to use with which DiscoHAT functions. But of course that only increases complexity, makes the board more complex (and less "robust"?) and makes the configuration of the software much more prone to error...

karrika, 2015-11-16:
Actually, http://pinout.xyz/ is a superb way for sharing information.

The only detail that I don't like is that the majority of board manufacturers solder their sockets on the board. This makes stacking HATs impossible.

DiscoHAT will be different. It has a SMD through hole connector allowing stacking of HATs. Of course it can only stack one board as nobody else seems to provide through hole SMD connectors. It is also obvious that the stacked board is an audio board of some kind.

RogueM, 2015-11-16:
To be fair, encouraging users to 'stack' their HATs when the specs clearly, and as I understand consciously, don't support it would be going against the point of the concept as I understand it's motivation to exist in the first place ;-)

lurch, 2015-11-16:
https://www.abelectronics.co.uk/default.aspx produce a range of stackable boards, but as you may suspect they're not HAT boards.

Gadgetoid, 2015-11-16:
It's not actually possible to have a stacked "HAT" since the auto-configure EEPROM would share only a single bus and address and the conflict would render them both useless. You'd have to explicitly make your board *not* a HAT and exclude the EEPROM so those pins can pass-through and allow the Pi to support anything stacked on top.

It's not made especially clear by the documentation, but I believe the intent of the HAT specification is *not* to support stacking at all, and you probably couldn't, in good conscience, use the "HAT" label on a board that does.

This doesn't really affect your design choices so much as your marketing and labeling. If you're keen to allow stacking, and don't mind potentially foregoing auto-configuration ( if/when it ever becomes a significant part of Raspbian for third parties ) then dropping the EEPROM can save a pretty penny off your BOM and RRP.

Unfortunately as you've already gone with the name "DiscoHAT" this could be problematic!

In your current setup, if I were to stack a SenseHAT on top of DiscoHAT then SenseHAT would simply not work and I, your confused customer, would be left scratching my head. Of course this can be fixed by adding `dtoverlay=rpi-sense` to `/boot/config.txt` but such complexity would go against the spirit of the HAT design.

The best information I can find on the topic is, basically: "Stacking is implicitly not supported for HATs, but you can design non-HAT stacking boards using the example surface-mount connector scheme"

What you should do is ask! And maybe consider "DiscoPi" ;D

lurch, 2015-11-16:
And don't forget to make it hamster-compatible ;-)
https://www.raspberrypi.org/blog/hamster-party-cam/

karrika, 2015-11-16:
Why do you maintain this pin usage information at all if you want to discourage people to add more than a single HAT?

Perhaps the HAT specs need to evolve. We have now only one eeprom sitting at address 0 (pins A0, A1 and A2 are tied to GND). By extending the address range by functionality you could easily support a bunch of HATs without a need for jumpers.

Example of categories (eeprom address pins A0..A2):
0 - default HAT that does not tell who he is.
1 - user interface HAT
2 - sound HAT
3 - light HAT
4 - motor HAT
5 - sensor HAT
6 - radio HAT
7 - hamster HAT

In this way you could stack any HATs as long as they belong to different categories.

lurch, 2015-11-16:
This was all discussed to death on the Raspberry Pi forums at the time the HAT specs were first announced...  try looking through the old posts on https://www.raspberrypi.org/forums/viewforum.php?f=100 and see also https://www.raspberrypi.org/blog/introducing-raspberry-pi-hats/

As already hinted at, there are plenty of add-on boards that **aren't** HATs.

(and just in case it's not clear, neither @Gadgetoid or myself were involved with the development of the HAT spec.)

RogueM, 2015-11-16:
> Why do you maintain this pin usage information at all if you want to discourage people to add more than a single HAT?

Pinout is not associated with the Raspberry Pi foundation. Combining add-ons is something made easier thanks to the info gathered here, true, but it's not the only meaningful purpose of the data.

But for what's worth, from my reading on the motivation behind the HAT specs it is to lower the barrier of use, and multiple EEPROM addresses would not prevent users to run into GPIO pin conflicts.

Personally I think of the HAT as a 'user-friendly' way to encourage people to expand on a barebone Pi without the frustration of never have learnt to walk before you try to run ;-)

karrika, 2015-11-16:
Thank you for all your comments. I know a lot more now. So tonight I make the last changes to the board and send it in for the first production run. I also ordered a sound HAT for trying out how the boards work together.

lurch, 2015-11-16:
Good luck with your product!

karrika, 2015-11-18:
Thank you. I believe I finally found a non-conflicting setup with most of the things I may need in the future.
One thing I am still missing is this "active": "high/low" thing. My interpretation is that it means when the signal is active. It could also refer to pullups/pulldowns. In my case I have external pullups so no internal weak pullups should be used.

{
    "name": "DiscoHAT",
    "manufacturer": "Kertatuote Oy",
    "url": "http://discohat.com",
    "description": "Computer controlled DMX lights, sounds and special effects.",
    "pin": {
        "8": {
            "active": "high",
            "direction": "output",
            "name": "DMX out"
        },
        "13": {
            "active": "low",
            "direction": "input",
            "name": "Button 1"
        }
        "15": {
            "active": "low",
            "direction": "input",
            "name": "Button 2"
        }
        "7": {
            "active": "low",
            "direction": "input",
            "name": "Button 3"
        }
        "18": {
            "active": "low",
            "direction": "input",
            "name": "Button 4"
        }
        "29": {
            "active": "low",
            "direction": "input",
            "name": "Button 5"
        }
        "31": {
            "active": "low",
            "direction": "input",
            "name": "Button 6"
        }
        "32": {
            "active": "low",
            "direction": "input",
            "name": "Button 7"
        }
        "36": {
            "active": "low",
            "direction": "input",
            "name": "Button 8"
        }
        "19": {
            "active": "high",
            "direction": "output",
            "name": "LED strip data"
        }
        "23": {
            "active": "high",
            "direction": "output",
            "name": "LED strip clock"
        }
    }
}

Gadgetoid, 2015-11-18:
Active high/low describes whether the device-side ( the HAT ) sees a low or high signal as being active.

For example, many devices will pull RESET up to VCC and expect a low/GND signal to trigger a reset. In this case the pin would be seen as "Active Low".

An LED may be connected directly to a pin and ground, in which case it would be "Active High."

Whether or not this information is remotely useful to Pinout users is debatable, and like pretty much everything else it was included on my whim with no real thought and no input from anyone else.

Right now, I don't believe there would be any ill effect from you simply omitting "active" altogether until we can figure out if it's useful and document it properly.

karrika, 2015-11-18:
Great! In that case I got it right. I already tested most of my DiscoHAT functions. The final boards will probably be here in 2 weeks. If DiscoHAT is accepted to pinout.xyz I make a pull request later.

thanks,
Karri

karrika, 2015-11-19:
I noticed that the Pinout2/overlay directory does not show up to me on the browser at all. Does it have some special permissions? I can only see it when looking at older commits.

lurch, 2015-11-19:
Are you looking for https://github.com/Gadgetoid/Pinout2/tree/master/src/en/overlay ?

karrika, 2015-11-19:
Nope. I am looking for the place where the JSON files are. They should be in Pinout2/overlay but it does not exist. At least not for me. So where is arduino-spi.JSON? Or actually when I do the pull request where do you want my discohat.JSON file.

lurch, 2015-11-19:
The JSON files are actually YAML files, and are in fact embedded within the MarkDown files in the directory I pointed at! Which you need to click on the "Raw" view to actually see through the GitHub web-interface, e.g. see https://raw.githubusercontent.com/Gadgetoid/Pinout2/master/src/en/overlay/explorer-hat.md

karrika, 2015-11-20:
Thanks. I just realized that I did not declare the eeprom pins, power pins and ground pins. They are all connected. But when I look at what other carts have been doing they have not declared them either. My assumption is that it is ok to leave them undeclared.

Or actually declaring the eeprom pins would be kind of important to distinguish real HATs from CAPs.

RogueM, 2015-11-20:
Yes, for now power/ground and EEPROM pins are not tagged on the pinout. We have several improvement planned and will amend the DiscoHAT overlay when that's done, no worries.

Thanks for your contribution!

lurch, 2015-11-20:
Yeah - see #52 and #64 for info. We've decided that the specific ground and power pins that a board connects to should be left out for clarity (apparently for some boards, different revisions of the same board connect to different ground pins!), but that the EEPROM ID pins **should** be included.
Also, at some point there'll hopefully be a new mechanism specifically for marking a board as HAT-compliant.

karrika, 2015-11-20:
lol

closed 2015-12-03 by RogueM