pinout-xyz/Pinout.xyz #63: Include / Display i2c address

lurch, 2015-11-16:
IMHO there should be a feature (i.e. built into the overlay 'data' itself, rather than just included in the 'markdown' part) for boards that "uses_i2c" to include the i2c address(es) the board uses. Which will be especially useful for non-HAT boards that use i2c and that can potentially be used together e.g. https://www.abelectronics.co.uk/
(although some of those boards include configurable i2c addresses, so I'm not sure (yet) what Pinout should do in those cases...)

RogueM, 2015-11-16:
yes. I think it is even there in some of the overlays, albeit as part of the Pimstaller experiment so may not be suitable. But either way, I agree, this info is pretty critical and does not need to be part of the translatable data.

... presumably there will be a factory 'default' address even if only whatever it might be without sobering tabs and/or jumpers? The sense HAT is another potential question mark, I wrote the overlay based on the data sheet of the components and that one *seems* to switch address depending on the state of the line, or maybe that of the interrupt pins. It's something I need to look into when I have time.

either way, I think the corner cases are probably not going to be hugely problematic, but something to keep in mind.

lurch, 2017-11-30:
Ahh, looking at https://pinout.xyz/pinout/adc_pi_plus (and https://raw.githubusercontent.com/Gadgetoid/Pinout.xyz/master/src/en/overlay/ab-adc-pi-plus.md ) I see that the I2C addresses are now a thing :tada: :+1: 

But looking at https://www.abelectronics.co.uk/p/56/ADC-Pi-Plus-Raspberry-Pi-Analogue-to-Digital-converter#assembly it seems that the I2C addresses are selectable, so it would be nice if https://pinout.xyz/pinout/adc_pi_plus could somehow display:
```
0x68 (or 0x6A, 0x6C, 0x6E): MCP3424
0x69 (or 0x6B, 0x6D, 0x6F): MCP3424
```
?

I also just spotted that https://pinout.xyz/pinout/di6acdq6rly_i2c_hat states that it's stackable, so perhaps the same distinction could be added to https://pinout.xyz/pinout/adc_pi_plus too? Ping @abelectronicsuk 

Or perhaps a boolean 'stackable' or 'stacking_header' value is worth adding as a standalone data attribute?

abelectronicsuk, 2017-12-03:
Hi

 

Thank you for the recommendation, I have created an updated listing all the I2C addresses. Do you think this way of listing the addresses will work ok or perhaps just have the extra addresses in the features section?

 

If this is ok, I will update the other boards.

 

Best regards

 

Brian

lurch, 2017-12-04:
@Gadgetoid and @RogueM will know more, but with the PR #246 that @abelectronicsuk has just pushed, I think pinout.xyz might assume that the ADC Pi Plus board uses all 8 i2c addresses at once?

I've got no idea how it'd work in terms of the YAML (I'm more of a JSON man), but I guess what might be nice would be for a board to support multiple "configurations"? So for this board you could specify something like:
`i2c_configs: [0x68, 0x69], [0x6A, 0x6B], [0x6C, 0x6D], [0x6E, 0x6F]`

And then what would be super-cool would be for https://pinout.xyz/phatstack (or similar) to recognise boards which have multiple i2c configs, and then allow you to choose _which_ i2c config you're using for a particular board, and then adapt the i2c ports used by a board as appropriate. Which would allow https://pinout.xyz/phatstack to stack 4 ADC Pi Plus boards without any conflicts being shown :-D

seamusdemora, 2021-06-17:
This is all great, but... I use the [RPi GPIO pinout](https://pinout.xyz/#) frequently, and assumed (until now) that all of the I2C buses would be identified as an `AltN` function when a pin is selected. But I don't see any of the other I2C pairs (SCL,SDA) identified here. Am I missing something - or is this information already "in the queue" for incorporation? - or is it tied in with #246 ?

lurch, 2021-06-17:
> I use the RPi GPIO pinout frequently

The `pinout` command included in [GpioZero](https://gpiozero.readthedocs.io/en/stable/cli_tools.html#pinout) _recommends_ the https://pinout.xyz/ website, but there's not actually any link between the two (entirely independent) projects.

> But I don't see any of the other I2C pairs (SCL,SDA) identified here.

What do you mean by "other I2C pairs"? The RPi only has two [I2C buses](https://pinout.xyz/pinout/i2c) - I2C0 (which is reserved for the HAT EEPROM), and I2C1 which addon-boards can use however they like.
(Technically there's also additional I2C busses on the Pi4, but **a)** #322 hasn't been fixed yet and **b)** it seems unlikely somebody would make an addon that is Pi4-only and doesn't work on earlier models of RPi?)

seamusdemora, 2021-06-18:
@lurch : I've depended upon https://pinout.xyz for years. 

RE **_"other I2C pairs"_**:  

I've currently got my RTC parked on `I2C1` (GPIO 2&3), but I need to free up GPIO3 to run some experiments with a "Start/Stop" button. I've read several sources ([example](https://medium.com/cemac/creating-multiple-i2c-ports-on-a-raspberry-pi-e31ce72a3eb2)) that state there are multiple I2C buses on the RPi4, and so I dug into `/boot/overlays/README` to see how to do this. I found the following overlays for the 2711 SoC in README: 

   * i2c3 
   * isc4
   * i2c5
   * i2c6
   *  i2c-rtc-gpio (a software controller  - no interest in bit-banging i2C, thank you)
   * i2c-rtc : which I'm currently using on i2c1, but could move to `i2c0` to free GPIO3 by specifying the `i2c0` parameter 

My current thinking is that the i2c-rtc overlay with the i2c0 parameter is my best bet (I have no HATs & no interest in them). I am under the impression that the README is *authoritative* wrt the overlays (and therefore many hardware assignments), but your post seems to be at odds with that?? 

I knew nothing of #322 until just now, but as I read it, it's a documentation issue - so I'm not sure how it affects what is in `/boot/overlays/README`.  

And finally - I'd like for you to clarify some of this if you could... Is my approach for moving my RTC to `i2c0` workable, or not? If not - do I have ***any*** workable alternatives?

lurch, 2021-06-18:
> @lurch : I've depended upon https://pinout.xyz for years.

Me too - and in fact I contributed some bug-fixes / improvements to the code behind [pinout.xyz](https://github.com/Gadgetoid/Pinout.xyz) in it's early days.
I've also contributed code to [GpioZero](https://github.com/gpiozero/gpiozero), including its `pinout` tool.

**The** _authoritative_ reference to the I2C busses in the Pi4 is the BCM2711 datasheet linked to from https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/README.md (which is a document I also did a large chunk of work on).

IIRC I2C0 is also used by the CSI (camera) and DSI (Touchscreen Display) interfaces (and possibly by the GPIO-expander used on some Raspberry Pi models?).

I've never experimented with the different I2C overlays (that you configure by editing `/boot/config.txt`) myself, so if the existing README doesn't answer your question, I guess your best bet is to ask for help at https://www.raspberrypi.org/forums/
But AIUI, yes if you're able to "activate" `i2c3` then you should / might be able to connect your RTC to that.

seamusdemora, 2021-06-18:
@lurch : Thank you for your contributions. 

But you've confused me - you said earlier that "The RPi only has two I2C buses", and now state that you've done considerable work on the authoritative reference for I2C buses. I'm sorry - I posted my question here hoping to learn if the I2C buses would be added to the document at some point, and then you indicated that there were only two I2C buses to document. OK - I suppose I am dealing in a *dark corner* here with the I2C bus documentation - I'll return to my experimentation. I don't mean to be offensive, and I hope it's not taken that way - I am confused by what you've said though. 

And I won't be consulting the forums as I was "banned for life" a couple of years ago. :))   Yeah - hard to imagine. LOL

lurch, 2021-06-18:
> But you've confused me

Sorry for causing confusion. When I said "The RPi only has two I2C buses" I actually meant "every Raspberry Pi before the Raspberry Pi 4".
I thought the "(Technically there's also additional I2C busses on the Pi4" made it clear that the Pi4 was different (to other Pis), but I guess I underestimated how complicated this is (which is unfortunately all to easy to do when you're working so near the coalface).

So:
* Raspberry Pi Zero, Raspberry Pi Zero W, Raspberry Pi Model A, Raspberry P Model B, Raspberry Pi Model A+, Raspberry Pi Model B+, Compute Module 1 - based on the [BCM2835](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/README.md) SoC - which uses the VideoCore IV - only two i2c busses
* Raspberry Pi 2B - based on the [BCM2836](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2836/README.md) SoC - which uses the VideoCore IV - only two i2c busses
* Raspberry Pi 3B, Raspberry Pi 3A+, Raspberry Pi 3B+, Compute Module 3, Compute Module 3+ - based on the [BCM2837](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2837/README.md) SoC (or [BCM2837B0](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2837b0/README.md) SoC for the '+' models) - which uses the VideoCore IV - only two i2c busses
* Raspberry Pi 4B, Compute Module 4 - based on the [BCM2711](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/README.md) SoC - which uses the VideoCore VI - six i2c busses

> I posted my question here hoping to learn if the I2C buses would be added to the document at some point

I'm not sure which "the document" you're referring to?

BTW, as it mentions in my [GitHub profile](https://github.com/lurch) I'm "only" a freelance software developer, so although I've done a bunch of work on various bits'n'bobs over the years (including for Raspberry Pi Trading), any information coming directly from an RPi Engineer should be taken as more authoritative than anything I've said :wink:

seamusdemora, 2021-06-19:
"coalface" :) that's funny - never heard that one before.

Yeah - misunderstandings... I should have more carefully considered the context here: this thread was started in 2015 - before the Pi 4B was created! I think your [reference to the BCM2711 datasheet here](https://github.com/Gadgetoid/Pinout.xyz/issues/63#issuecomment-863843176) was what  caught me sideways. 

And "the document" is the document served up at the [pinout.xyz website](https://pinout.xyz/#) - the subject of this repo, and this issue. 

And finally, FWIW, I've always felt that being outside an organization gives one a clearer understanding of things. Oftentimes, an asset. 

Thanks for clearing that up.

closed 2026-08-18 by Gadgetoid