pinout-xyz/Pinout.xyz #70: Strange navigation sequence

lurch, 2015-11-18:
If I go to http://pinout.xyz/ and then click on the French flag, it takes me to http://fr.pinout.xyz/
If I then click on the "Raspberry Pinout" logo in the top-left corner, then it takes me to http://fr.pinout.xyz/pinout and all the flag icons at the top right suddenly disappear!

And similarly, if I manually browse to http://pinout.xyz/pinout/ I get a page where the flag icons display the 'broken image' icon, and if I manually browse to http://pinout.xyz/pinout (or http://fr.pinout.xyz/pinout/ ) I get "**Not Found**".

IMHO it'd make sense for http://pinout.xyz/pinout and http://pinout.xyz/pinout/ to redirect to http://pinout.xyz/ , and for http://fr.pinout.xyz/pinout and http://fr.pinout.xyz/pinout/ to redirect to http://fr.pinout.xyz/ .
Or if that's not possible, maybe the page that gets displayed at http://pinout.xyz/ should also get automatically copied to http://pinout.xyz/pinout and http://pinout.xyz/pinout/ .

OTOH (and at the risk of breaking any bookmarks that people may have made), is having everything in a 'pinout' subdirectory (e.g. http://pinout.xyz/pinout/pin5_gpio3 and http://pinout.xyz/pinout/uart and http://pinout.xyz/pinout/sense_hat ) even useful in the first place? I guess we could either strip out the subdirectory (e.g. http://pinout.xyz/pin5_gpio3 and http://pinout.xyz/uart and http://pinout.xyz/sense_hat ), or separate the files up by using more logically named subdirectories (e.g. http://pinout.xyz/pin/pin5_gpio3 and http://pinout.xyz/bus/uart and http://pinout.xyz/board/sense_hat ) ??

Gadgetoid, 2015-11-18:
Dropping or replacing the /pinout subdirectory is on my todo list. But as you rightfully point out, it's a hell of a bookmark breaker. I'd have to set up some 301 redirects to get everything back on track. Admittedly I should have done it as part of the pi.gadgetoid.com/pinout -> pinout.xyz migration, but I didn't want to cause myself any headaches I could avoid by being lazy.

There shouldn't be any links to /pinout, these will be some legacy fallout from when the root URL was pi.gadgetoid.com/pinout. It looks like a pinout.html page has leaked through into some of these folders, though, since fr.pinout.xyz/pinout shouldn't even work. I've deleted it!

I'll have to set up some redirects in the server configs to ensure nobody ends up in 404 oblivion.

lurch, 2015-11-18:
> I'd have to set up some 301 redirects to get everything back on track.

If you're using Apache, that's easily solved with a quick [mod_rewrite](https://httpd.apache.org/docs/2.4/rewrite/remapping.html) rule.
But as I noted in #35 the URLs of the translated pages will change anyway, as their "name" gets translated from English to the local-language equivalent. (e.g. http://fr.pinout.xyz/pinout/controleur_moteur_ryanteck used to be at http://fr.pinout.xyz/pinout/ryanteck_motor_controller_board - oh my, the old page actually still exists too!)

> It looks like a pinout.html page has leaked through into some of these folders, though, since fr.pinout.xyz/pinout shouldn't even work. I've deleted it!

I guess these pages also "shouldn't work" ?
 * http://pinout.xyz/pinout/
 * http://de.pinout.xyz/pinout/

For URLs that don't exist, maybe it'd be more friendly to provide a link (or automatically redirect) to http://pinout.xyz , rather than just displaying the default "Not Found" ?

> Aha, here's the offending line in the French template: That URL should probably be auto-generated, or just be "/"

I agree that URL ought to be auto-generated (in all templates), similar to the way that `{{resource_url}}` is used.

Gadgetoid, 2015-11-19:
As part of the response to general navigation weirdness I'm experimenting with proper 404 handling, using a useful page rather than the default mess. See: http://pinout.xyz/pinout

This is a little hacked-up at the moment, and the 404 would need generating as part of generate-html.py and, of course, translating. Another thing for the TODO list! Should be relatively easy, however, and injected the same way the index page is.

lurch, 2015-11-20:
Great :+1: 
The http://pinout.xyz/pinout/ address I mentioned in my previous comment still exists though (with the broken flag images), whereas I guess it too should 404 now?

And I also guess the "hacked-up-ness" is why it doesn't work for any other version of the site than English yet? ;)

Gadgetoid, 2016-02-17:
I've finally deleted the rogue http://pinout.xyz/pinout/ and fixed the 404 following my breaking it horribly with the recent visual change. The 404.html is currently generated manually, which doesn't help matters.

RogueM, 2016-11-03:
closing... not sure any of what's been discussing is addressed, but it's a tad difficult to assess whether there's still any relevance, the site having changed so drastically in the last 11 months.

closed 2016-11-03 by RogueM

lurch, 2016-11-03:
Agreed - and the 404 pages are much nicer formatted now :)