pinout-xyz/Pinout.xyz #82: Add raspberry pi cobbler pinout

m4recek, 2016-01-07:
Hi, lot of people have bought this cobbler board for raspberry pi, can you please include pin numbers in your diagram?

http://i.imgur.com/snSdpaO.png

RogueM, 2016-01-08:
Pinout is really for add-on boards. It's unfortunate that this cobbler label is less than helpful. Note also that it's not correct to use a 'GPIO' notation to describe physical pins, those are completely different identification systems.

m4recek, 2016-01-10:
Printed names on that cobbler are actually names of GPIO pins from this table http://wiringpi.com/wp-content/uploads/2013/03/gpio1.png its from wiring pi website http://wiringpi.com/pins/

> Note also that it's not correct to use a 'GPIO' notation to describe physical pins
I believe thats what Adafruite library does already... so its used in a lot of projects, or did you ment something else?

lurch, 2016-01-10:
If all you need is the WiringPi pin numbers, then have a look at http://pinout.xyz/pinout/wiringpi  :-)

m4recek, 2016-01-10:
> If all you need is the WiringPi pin numbers, then have a look at http://pinout.xyz/pinout/wiringpi :-)

No, I need names of the pins as on [linked image](http://wiringpi.com/wp-content/uploads/2013/03/gpio1.png) from WiringPi website and their relation to BCM and WiringPi numbers... I had that image printed now so its not a problem for me anymore. 

But you can save other peoples time, its quite hard to find.

lurch, 2016-01-11:
I'm sorry to sound blunt, but the "Name" column on that WiringPi image is an outdated labelling scheme, and IMHO it shouldn't be encouraged. Maybe if you have a fine-tip pen (or sharp pencil) you could put a sticker on your existing 'cobbler' board, and label the pins with either the BCM numbers or the physical pin numbers (whichever you prefer)?

From a quick google-search, there appears to be lots of different 'cobbler' / 'breakout' boards for the Raspberry Pi, so I don't think Pinout2 could ever hope to cover all the varieties.

On a side-note, it's unfortunate that http://wiringpi.com/pins/ *still* hasn't been updated for the B+ / Pi2 40-pin header :-/

m4recek, 2016-01-11:
Its only naming scheme not showed in Pinout that I know about :) Other cobblers use BCMs. 
People wont stop using it just because its old, and you can still mark it as old in naming scheme list.

So my points for showing this naming scheme, under other naming schemes are:
1. Lot of people bought them already
2. Its still sold in shops 
    - random shops around the world [in uk](https://digitalmeans.co.uk/shop/breakout_kit_for_raspberry_pi_to_breadboard), [in germany](http://www.avc-shop.de/GPIO-KIT)
    - [amazon](http://www.amazon.de/T-Cobbler-GPIO-Raspberry-Breadboard-Steckboard/dp/B00F7PRJN4)
    - ebay, aliexpress...
3. Its used in some older raspberry howtos.
4. People (especially newcomers) are having problems with it
    - me now
    - [this guy](https://www.raspberrypi.org/forums/viewtopic.php?p=509634) a year ago
    - [this guy](https://www.raspberrypi.org/forums/viewtopic.php?t=64488&p=475613) year before that
    - and a lot more... try to google that image

Pinout is referenced in a lot of places for solving pin reference on raspberry. So its natural that newcomers will eventually find it.

Gadgetoid, 2016-01-28:
Pinout actually tells you WiringPi pin numbers on mouseover in the tooltip text. Useless on a tablet or phone, but it's a sneaky hidden bonus if you know where to look ;)

The trouble with the titles under the "name" column is that they're a mix of WiringPi GPIO assignments and the names of some pin functions.

An overlay could easily be created with this names for reference, but it would just be presenting information that's already in Pinout and easily determined by tabbing between http://pinout.xyz/pinout/wiringpi and http://pinout.xyz

closed 2016-02-17 by Gadgetoid