pinout-xyz/Pinout.xyz #83: Sense HAT GPIO assignments are incorrect

P33M, 2016-01-11:
https://www.raspberrypi.org/wp-content/uploads/2015/08/Sense-HAT-V1_0.pdf

The pin assignments on the site do not correspond to the actual functional implications.

SPI { pi, Atmel } is not used by default. It is used for production test and programming the Atmel. The framebuffer is driven via I2C.

The joystick is not connected to { 23, 24, 25 }. It is sampled by the Atmel and accessible via the I2C interface.

lurch, 2016-01-11:
Could a link to that PDF be added to https://www.raspberrypi.org/documentation/hardware/sense-hat/README.md ?

Gadgetoid, 2016-01-28:
Sense HAT is odd- I never understood why it was engineered the way it was. It's brilliant from an engineering/software perspective, of course, but depending upon mysterious keyboard and display drivers existing and functioning on the Pi is a curious choice when a simple dumb library could easily read inputs and drive an output display. I guess it's to do with reliability in space?

I tested a Sense HAT once with a broken EEPROM and it basically didn't work because all the magic glue wasn't loaded...

Aannnyyyway! This clearly needs looking into!

RogueM, 2016-01-28:
yes, I didn't have a board when I made the overlay... clearly wrong. I *think* I have a revision on my local machine, I'll have to check.

P33M, 2016-01-28:
"It's a 2-layer board with 70% of its area devoted to an LED matrix" is pretty much what drove the design choices. Software is an amortized cost over N sales. Increasing the complexity of a board is a fixed cost per-board.

Gadgetoid, 2016-02-17:
The display setup is pretty phenomenal, and of its idiosyncrasies it's definitely the more technically impressive. You don't know what people are likely to do with a ridiculous frame-rate until they do it.

It's undoubtably what happens when people who- "know what the hell they're doing"- design a Pi add-on.

closed 2016-06-25 by RogueM