pinout-xyz/Pinout.xyz #84: Add Carberry-Board

mboremski, 2016-01-11:
There is a nice board to use your PI in your car.
It includes a DC/DC-converter, connections to CAN, LIN and so on.
Could be interesting to add it to the boards.

Web: http://www.carberry.it/
Pinout: http://www.carberry.it/wiki/carberry:hw_spec:raspberry_pi_interface

Gadgetoid, 2016-01-28:
I think most of us are experiencing something of a time crunch at the moment ( as you can tell by it taking 17 days to even reply :D ), so the best way to get a board added quickly is to fork the repo and have a go yourself.

Anyway, it's duly noted for consideration if I get some time!

RogueM, 2016-07-09:
this is done in the repo, to the best of my ability and brief research into the board info... @Gadgetoid will push live at some point soon. If you wish to modify/correct any of the available info, please submit a PR tweaking the carberry.md as required.

closed 2016-07-09 by RogueM