pinout-xyz/Pinout.xyz #85: Should we reinstate which VCC / GND pins a board might use, if known?

Gadgetoid, 2016-01-28:
It was never really an issue with HATs, but the headerless pHAT format opens up a variety of different ways you might connect one or more boards to the Pi.

I think knowing which VCC / GND pins a board uses ( since they may not always be consistent ) is useful, although I do agree with the original reasons for removing them- they're visually noisy in some cases.

So I'll put it to discussion. Should we reinstate them? And if so, how could they be implemented subtly.

RogueM, 2016-01-28:
Pimoroni use sensible routing, pin 1(3)/2 for VCC (with only 1 exception as dot3k IIRC). GND can be tricky sometimes though (and would be a nightmare to document)... but sure, something is needed - I had started some work towards that goal pre-xmas ;-)

RogueM, 2016-01-28:
here's the commit:
https://github.com/Gadgetoid/Pinout2/commit/e352c73d35f5de6130158f1e22108e102cb4045f

... of course it does not map VCC to specific pin, but personally I think it should assume pin 1 and 2 for 5v and 3v3 respectively, the exceptions can be handled at a later stage.

ground is a big can of worm, I really would not want to touch it with a barge pole, not just for the noise it creates on the overlay but also the oddities I have ran across in testing... again assuming any and all (or perhaps pin 6 by defect?) would be my preference.

Gadgetoid, 2016-02-10:
A good, recent example of the complexities with Ground is the Pimoroni Unicorn HAT. It needs a *specific* ground pin to be connected, which has a dedicated trace routed to it. This, and any board without ground fill, would be baffling to users who didn't know the secret sauce.

If there weren't so many bloomin' ground pins, I'd just say: "Assume all ground connections are required."

RogueM, 2016-02-10:
It's hardly an exception, I've started mapping those for all the boards I have, meaning mostly Pimoroni. 

I'm too ridiculously unskilled to modify the code to tag the overlays appropriately though - I really don't want to start re-crowding the overlays with power/ground undistinghished at a glance from the GPIO/function pins though.

But the problem is going to be to get accurate info on all contributions though, which is why I think a sensible 'default' behaviour is required, yielding a readable overlay, otherwise we'll be back to square on.

Gadgetoid, 2016-02-16:
Yeah you're not wrong regarding accurate info- in most instances ( short of exhaustively testing every new board) the only place we're going to get that is from the board designer. Easy for all Pimoroni boards because I can double-check the schematics and routing- a bit of legwork and nudging for others.

I like the idea of small, subtle pips indicating which specific pins are essential, along with your suggestions for a more general "uses 3v3" and "uses 5v" indication.

closed 2016-09-27 by Gadgetoid

lurch, 2016-09-27:
I noticed that pinout.xyz *does* now do this, but the Power and Ground 'blobs' obscure the pin numbers (e.g. 2 and 6 on http://pinout.xyz/pinout/blinkt ), due to them being the same colour. Could the pin-numbers be shown reverse-video-style (i.e light on a dark background) in this particular case?

RogueM, 2016-09-27:
I tried that solution but was not able to get a pleasing result, the blobs would have to bleed deeper and the result was rather messy.

Overall I decided the loss of pin number for ground (and power) was not a huge deal, with neighbouring pins still implying the numbering, and besides physical for ground has no real use (except perhaps for overlay makers).

To cut a long story short, after discussing briefly with @Gadgetoid who had pointed as much, I think we both agreed that was fine, and I've certainly came to appreciate the (accidental) emphasis being on pin numbers that actually matter for users (whether they prefer to work with BCM or physical scheme).

lurch, 2016-09-27:
Fair enough :) :+1: