pinout-xyz/Pinout.xyz #89: Board Categories

Gadgetoid, 2016-02-22:
Still debatable, but I'm thinking something like:

* Robotics & Motors
* LED & Lighting
* Displays
* Audio & Video
* Input & Controls
* Sensors
* Microcontrollers
* Other

I'd also love to get images for every single board because they look so good as thumbnails, but that's going to be a honking great amount of work!

lurch, 2016-02-22:
Sounds reasonable. What would be covered under "Video" that doesn't come under "Displays"? (i.e. should "Audio & Video" be just "Audio"?)

Would the DrumHAT and PianoHAT be "Audio" or "Input"? ;-)

Maybe there could be a "Prototyping" section for things like the Explorer & Propeller HATs?

It might make sense to keep the categories fluid (i.e. easily changeable over time) so that the "Other" category doesn't end up getting over-populated?

The thumbnails on http://pinout.xyz look nice, but as the board-categories get populated, the images might end up making the menus too big? No reason that the thumbnails couldn't be displayed on the board-specific pages though?

Sorry I've not been very active on this project after being so eager initially, but https://github.com/RPi-Distro/python-gpiozero is now taking up most of my spare-time-hacking focus!

P.S. Might be useful to put a link to http://pinout.xyz on https://github.com/Gadgetoid/Pinout2 where it currently says `No description or website provided.`

Gadgetoid, 2016-03-23:
No apology needed, GPIO Zero is a noble cause, which I'd honestly much rather you contributed to than this backwater mess of a website ;) ( not that I don't wholly welcome your input )

I've added more thumbnails recently, and at some point we're definitely going to run into the limitations of the new pop-down menu setup. I'm all for moving away from a crazy menu into individual category pages where boards are listed with thumbnails and description.

Or... perhaps a single page with filtering/sorting where *all* boards are listed, and easily explorable. From here, functionality such as "Does this board work with this board" could potentially be explored.

Food for thought in a Pinout 3.0... perhaps.

Gadgetoid, 2016-06-26:
Since the structure of and approach to this problem has changed so dramatically, I'm closing this issue so that any new concerns can be discussed elsewhere!

closed 2016-06-26 by Gadgetoid