pinout-xyz/Pinout.xyz #91: Adafruit Capacitive Touch HAT

prjemian, 2016-03-13:
Just came across your GPIO pinout page display (sounds scandalous) and was hoping there might be some help for the Adafruit Capacitive Touch HAT.  This device sounds wickedly fun, especially when used with the 7" touch screen interface.

Web: https://www.adafruit.com/products/2340
pinout: https://learn.adafruit.com/mpr121-capacitive-touch-sensor-on-raspberry-pi-and-beaglebone-black/hardware

I'm aware of your backlog note, https://github.com/Gadgetoid/Pinout2/issues/84#issuecomment-176342666, but add this in case y'all get some time.

Gadgetoid, 2016-06-16:
The reason there are so few other boards on, aside from Pimoroni products that is, is mostly because I don't have access to them to be 100% confident of the pinout. But this is definitely one of the #1 things I want to fix about the site.

I'll see what I can do!

lurch, 2016-06-16:
Maybe Pimoroni could provide a bit of "sponsorship" to Pinout, and buy you all the add-on boards that you don't have access to? ;-)
(But of course that's irrelevant if you want to keep Pinout 100% independent)

RogueM, 2016-06-25:
overlay submitted... as noted no access to board puts a limit on accuracy but I'm fairly confident there is no major mistakes.

closed 2016-06-25 by RogueM