pinout-xyz/Pinout.xyz #94: The website is somewhat difficult to use on mobile

cgsmith, 2016-06-09:
This might just be me, but when on a mobile the website is fairly difficult to use. I'd love to help out or offer feedback to the project where necessary.

RogueM, 2016-06-09:
I tend to use it on a tablet and it works well... on what device are you and are there particular problems you could describe?

cgsmith, 2016-06-09:
It is difficult to click or see all the pertinent information. I am using a
Moto E with Android.

RogueM, 2016-06-09:
I guess it depends how you use the site. I tend to select my board then zoom in the pinout on left, as I don't really need the descriptions. I can imagine that if you want both visible you'd be squinting a fair bit.

Gadgetoid, 2016-06-16:
This is fairly low priority, but I'd like to have a go at making a mobile-friendly version of Pinout at some point. Possibly even in "App" form- although that may be a web App using a JS framework, rather than something native.

I'd achieve this by displaying only the Pinout part of the detail, and moving the text description/extra info into an additional screen.

Google likes mobile optimized, too, and Webmaster Tools complains about the poor display of Pinout on mobile.

Gadgetoid, 2016-11-01:
For now, I don't think that shoe-horning this density of information into the size of a mobile-phone screen is possible.

I leave the door open for the possibility of an App one day, but for the website it's easier to wait for mobile phones to catch up than waste time trying to accommodate them now.

closed 2016-11-01 by Gadgetoid