pinout-xyz/Pinout.xyz #97: RasPiO-Analog-Zero

francesco-vannini, 2016-06-26:
Pinout is standard SPI. So mosi,Miso,sclk & CE0
http://rasp.io/wp-content/uploads/2016/06/RasPiO-Analog-Zero-graphic-bare.png

Would you need something more to add this board?

RogueM, 2016-06-26:
Hi Francesco! in an ideal world we'd prefer submissions to be via pull requests to the repo in the form of an associated markdown file... even if it's not perfect it would save us a lot of time.

It's not too difficult, really, all you need is a text editor and to grab the overlay template from here:
https://github.com/Gadgetoid/Pinout2/tree/master/src/en/template

Fill in as much as you can in the top half (use 'not known' if necessary for fields you are uncertain about), and the bottom part with a useful shortish description i.e the essential about the board, visitors (and us) can refer to the URL you submit in the top half of the overlay for further details.

Don't worry about getting everything right though, we can check the data will work in the framework of pinout, but cutting out on research time at our end will ensure board requests don't merely accumulate week after week after week - unfortunately our time is limited, as the burst of activity here spread out in time should attest.

If you feel that what I outline above is beyond your possibilities, then contacting the board manufacturer and pointing them to this site may encourage them to make board submissions themselves, which benefits them, us, and visitors i.e everybody wins!

Thanks for your support - either way, we'll look at your couple of requests, but it may take a while before we are able to do so!

Gadgetoid, 2016-06-26:
I second the above message. We're pretty stacked with the architectural considerations of Pinout, and  don't have as much time as we'd like to spare adding tons of new boards. That's the main reason for the lack of them thus far.

Pinout gets something in the order of 100k page views a month, with 20k of them being on individual board pages- and we're improving the site to facilitate better discovery of new boards with a faceted search facility. For anyone submitting their own boards, it's going to be well worth the effort!

Thank you for making it this far, but a pull-request would make us very happy bunnies! :D

RogueM, 2016-07-09:
this is done in the repo, @Gadgetoid will push live at some point soon.

closed 2016-07-09 by RogueM