pinout-xyz/Pinout.xyz #106: Incorrect/inconsistent voltages for Pi and Arduino

awmartin, 2016-07-16:
On [this page](https://github.com/Gadgetoid/Pinout2/blob/master/src/en/overlay/uart.md), the page starts by stating the _Pi has a 3.3v line_ and the Arduino, _5v_.

Then, in the second paragraph, the text states converting the _Pi's 5v_ line to the _Arduino's 3.3v_ line with a voltage regulator.

I believe the first statement is false.

RogueM, 2016-07-16:
No, there's no error there. Basically what @Gadgetoid is saying is that you can run an AVR at 3.3 logic and if that's what you want to do dropping the 5V line of the Pi to 3.3 is a handy way to achieve this, with enough stability for the crystal not to miss a beat.

As to whether the discussion about Arduino is particularly relevant to the UART description page, it's a matter of opinion. In mine, it's more something inherited from the original Pinout site and probably should be reviewed. So I'll leave this open and discuss with Phil to see what, if anything should be done.

awmartin, 2016-07-17:
I see. This was not clear to me from the wording that the Pi had two voltage outputs, and the Pi's 3.3v output could cause instability in this use case with the Arduino being able to run at two voltage levels as well. I suppose this page is meant for a more advanced reader. I'm not even sure what "AVR" refers to from your comment. My apologies.

closed 2016-07-17 by awmartin