pinout-xyz/Pinout.xyz #107: Typo(s) in german Translation of pinout/pin5_gpio3

pxlphile, 2016-07-25:
Hi,

thanks for your service providing GPIO layouts in such a wonderful way. Too bad I found a spelling error in the german page https://de.pinout.xyz/pinout/pin5_gpio3:

"SCL ist der Clock oder Tackt Anschluss des I2C-Bus des Pi. mehr über I2C."

It shouldn't say "Tackt" but "Takt" which is the same word for the music beat. For further correction of compount words (incl. Omission dash) and uppercase of the sentence's first letter of the second sentence I propose the following (changes shown with markdown):

"SCL ist der Clock**-** oder **Taktanschluss** des I2C-Bus des Pi. **Mehr** über I2C."

Thanks :)

RogueM, 2016-07-25:
Hi - cheers for pointing that out. Best way is to submit a PR, particularly for translations as we don't speak German so us tweaking them could result in a worst result.

... That said this one looks straightforward enough so let's see what happens when I put these words of wisdom said ;-)

RogueM, 2016-07-27:
Fixed - will go live next update.

closed 2016-07-27 by RogueM