pinout-xyz/Pinout.xyz #108: Missing url to site in Portuguese

ifribeiro, 2016-07-26:
When I click in the Portugal's flag, the page should changing the language to Portuguese, but this don't happen. Then I realize that the error was just the 'pt' missing after the "http".
e.g: http://pinout.xyz/ should be http://pt.pinout.xyz/

RogueM, 2016-07-27:
We'll look into it... unfortunately the Portuguese translation is very much a draft - we had someone starting on it but only submitted the bear minimum.

We are currently reviewing all the translation portals to make them compatible with the new layout with board filtering at that point, so we'll assess if PT should come down (given the limited value currently) or stay in.

In the meantime, if you would like to contribute you can grab any of the files located at https://github.com/Gadgetoid/Pinout2/tree/master/src/pt/translate

Any help is appreciated. If you have no interest yourself contributing, then sharing the site with members of your community with a note to the extend that we are looking for translators would go a long way to take pt.pinout.xyz to its full potential ;-)

ifribeiro, 2016-07-27:
I forgot to say before, the site has been very helpful for me. Thank you! And I have to say that you guys are doing a great job with it. Congratulations :-)

I will try translate what I can (my English is not that good), but first I need to learn how to work here in GitHub. :-)

RogueM, 2016-07-27:
It's relatively easy, the hardest part is creating an account and you've got that covered already ;-)

Just fork this repo, then modify the .md files in the Pinout2/src/pt/translate/ folder. Then it's just a matter for you to submit a PR and us to review and publish it.

The part of the md files that need to be translated is mostly located below the line in the format:
#board-name (about half way down the files typically).

There's a few more info that needs localising in the header, most notably the <description> field though. If you like you can simply include it directly below the '#' line preserving the word description at the front and we'll sort it out. Not that it's difficult to translate in place but there may be a few pitfalls when modifying tags.

... even if you left the <pin> untranslated I think the result would still be immensely more useful than an untranslated page, so don't worry too much about any aspect that may seem like an obstacle at first, give it a go and we can give you feedback on how to take it to the next step once you've gained confidence in the process involved.

ifribeiro, 2016-07-28:
I have tried some translations... it's seems to be easy. Thanks for the tips :-)

RogueM, 2016-10-22:
we're killing the pt placeholder for now. If you (or someone else) want to pick the pt translation up again in the future, just duplicate 'en' into a new 'pt' folder, preferably rename the 'overlay' dir to 'translate' and make PR with your translated overlays.

closed 2016-10-22 by RogueM