pinout-xyz/Pinout.xyz #115: Graphical Pinout

lurch, 2016-09-27:
Just spotted the new graphic on http://pinout.xyz/ - nice! :+1: 

Just a couple of little suggestions, which you're obviously free to ignore if you wish:
  * IMHO the pink colouring on the BCM pins 8, 7 & 19 should extend outwards slightly to also cover the 'CE0', 'CE1' and 'MISO' labels. And the same for 'ID_SD' on BCM pin 0.
  * I think the diagram should definitely make clear that the SPI interface on BCM pins 10, 9, 11, 8 & 7 is actually different from the SPI interface on BCM pins 19, 20 & 21, and the pins from each can't be intermixed. Perhaps change the labels to SPI-0 and SPI-1 (or similar)? And maybe there should be a "fatter corner" (IYSWIM) on the pink blobs between BCM19 and BCM20 to make it clearer that they're a part of the same SPI block?
  * It might also be worth doing something similar to differentiate I2C-1 (BCM 2 & 3) from I2C-0 (BCM 0 & 1)
  * Perhaps the GND pins should have a white dot in the middle, like the diagram on the left of http://pinout.xyz/ has, rather than being completely black?
  * Could you get rid of the very thin pink line between BCM8 and BCM11? ;-)

Gadgetoid, 2016-09-27:
Agreed on all counts :D Thanks for your feedback. I've pushed the updated version up, along with the source .ai file in the interest of continued Pinout openness.

[image]

RogueM, 2016-09-27:
shouldn't physical 27-28 be I2C0 though?

Gadgetoid, 2016-09-27:
Yeah I've now fixed that :D and a couple of other wee tweaks.

lurch, 2016-09-27:
I'd be tempted to keep the spaces between the bus-names and bus-numbers i.e. "I2C 0" instead of "I2C0", but other than that it's all good. Thanks for taking on board my suggestions :-)

closed 2016-09-27 by lurch

kevinwchang, 2016-09-29:
Two more issues I noticed:
- BCM 20 should be MOSI, not MISO.
- The two 3.3 V pins would be more accurately labeled 3V3 instead of 3V.

Aside from those, thanks for the nice diagram!

RogueM, 2016-09-29:
good spot!

Gadgetoid, 2016-09-30:
Updated the `.ai` to include the corrections, re-exported to 72 an 300dpi - good idea @RogueM - have also re-exported to `.eps` which I hopefully haven't borked up. I think it's a much better format for a creative-commons graphic to be available in, since not everyone can deal with `.ai` gracefully.

RogueM, 2016-09-30:
SVG would be even better, but I could not generate it from the AI without dropping objects.

lurch, 2016-10-01:
Obligatory https://xkcd.com/927/
Sorry ;-)

Sounds like a job for a neat little Python Pillow script though.