pinout-xyz/Pinout.xyz #117: Mouseover area is wrong for odd pins

lurch, 2016-09-29:
Something I just noticed - don't know if it's been like that for ages, or is only a recent problem...

On http://pinout.xyz if I mouse-hover over "BCM4" then BCM4 and physical pin 7 get highlighted. If I mouse-hover over "BCM14" then BCM14 and physical pin 8 get highlighted. If I mouse-hover over the pin-number "7" then BCM4 and physical pin 7 get highlighted. If I mouse-hover over the pin-number "8" then BCM14 and physical pin 8 get highlighted. If I mouse-hover over the purple circular-pin next to the pin-number "8" then BCM14 and physical pin 8 get highlighted.
All as expected.

But if I mouse-hover over the green circular pin next to the pin-number "7" then BCM14 and physical pin 8 get highlighted!  Tested in both Chrome and Firefox on Ubuntu 14.04 - both behave the same (for all the circular pins on the left-side of the pinout header).

RogueM, 2016-09-30:
Cheers, I'll have a look tonight... could totally be the result of my fudging to style the power and ground pins.

Gadgetoid, 2016-10-07:
I think I've fixed this now, and made a few other CSS tweaks to neaten it up and change some things that were bugging me :D

See: http://pinout.xyz/pinout/display_o_tron_hat

* Dropped transparency of unused pin labels to make it less busy (hopefully)
* Reintroduced physical pin numbers to Ground/Power nubs
* Added mouseover effects to Ground/Power nubs
* Added slight mouseover highlight to used pins
* Made the 3v3 power (pin 1) mouseover right border radius square to match the pin

Also in a previous update I sneaked the font size up a bit. I'm always doing this if I can get away with it. I have a lot of 10px to 12px font usage to atone for!

The details section now includes:

* Whether 5v, 3v3 or both power supplies are required
* If SPI is used

closed 2016-10-07 by Gadgetoid

RogueM, 2016-10-07:
I think the labels for used ground pins shouldn't be that light, perhaps. On the other hand I don't want to give the impression they are ALL required so I'm not entirely sure what's best.

oh, yeah... there's also some earlier code I'd thrown in that is supposed to display in the details section whether there is an EEPROM, but I obviously botched that job since it doesn't :/