pinout-xyz/Pinout.xyz #134: No flags at the bottom of http://pinout.xyz/boards

lurch, 2016-10-19:
When I'm looking at http://pinout.xyz/ there's lots of flags along the bottom, which I can use to switch language. But when I go to http://pinout.xyz/boards there's no flags at the bottom, which means if I want to switch language, I have to go all the way back to the home-page, and then re-navigate to e.g. http://de.pinout.xyz/boards

Gadgetoid, 2016-10-22:
D'oh! Good spot. They will be missing from the template file, or at least the placeholder to include them. I shall take it upon myself to fix this if nobody beats me to it :D

RogueM, 2016-10-23:
The subdomain content system I implemented last night make this issue largely unimportant, but I think the flags should nonetheless be spliced in the board template... I'll have a look and see if I can blag my way through that one too ;-)

RogueM, 2016-10-23:
hum, no, the class is in the template, but the rendered board.html has a zero population list of lang... I'm not clear on what the problem might be.

RogueM, 2016-10-23:
I think I traced down the problem to the get_lang_urls method in the html generator. The problem is that function won't return any urls.

For now made the lang navigation on the boards page the same as the index page, meaning it will return the user to the top of the relevant subdomain.

... so, not quite the proper fix, but I'm not sure it's worth the hassle to refine the redirecting. Feel free to do so however, if you think it's important (in the context of switching lang being much lesser relevant now).

Gadgetoid, 2016-10-28:
I think it's safe to assume in all cases that the lang version of /boards:

* Exists
* is lang.pinout.xyz/boards

I should be able to fudge up get_lang_urls to account for that :D

RogueM, 2016-10-28:
yes, that's essentially it. I haven't really got a clue why the live site doesn't show flags for all lang on each page though, after the latest tweaks it should (and did while I was testing, though they would take me to the live site of course).

RogueM, 2016-10-28:
hum, no, I must have dreamt it, offline serving yields the same result as what is live now. Culprit is likely in the same function I suspect.

Gadgetoid, 2016-11-01:
Fixed the boards flags in ba08c9aa34bfb32ea79db14794915af0f2b1e4b1

And, hopefully, the flags in 4ddc00df90e06816b5accef1d4d25177f865aefc

lurch, 2016-11-02:
Getting closer, but still not _quite_ working properly...
  * When I'm on pinout.xyz/boards the flags at the bottom link to e.g. http://fr.pinout.xyz/boards (correct), but when I'm on http://fr.pinout.xyz/boards the flags at the bottom just link to e.g. http://de.pinout.xyz/ (incorrect, should be http://de.pinout.xyz/boards )
  * I get all the flags at the bottom of http://fr.pinout.xyz/pinout/traffic_hat (correct) but only _some_ of the flags at the bottom of http://fr.pinout.xyz/pinout/witty_pi_2 (incorrect?)

RogueM, 2016-11-02:
Yes, I don't think anything but en was republished last night, so the situation would be unchanged on the live site. I think.

RogueM, 2016-11-02:
I just checked a local build and _I think_ it's indeed fixed.

lurch, 2017-11-30:
Yup, just done some testing and seems all good now, sorry I forgot to close this issue.

Would there be any way for the flags at the bottom to also work with the query-string part of the URL?
E.g. if I'm on https://fr.pinout.xyz/boards#manufacturer=Adafruit and I click on the Italian flag, it would be cool if it then changes to https://it.pinout.xyz/boards#manufacturer=Adafruit - at the moment it just changes to https://it.pinout.xyz/boards
This is only a very minor problem though, not worth re-opening this ticket for.

Gadgetoid, 2017-11-30:
Actually, yes that should be possible with a little Javascript that appends the right querystring to that URL. Good idea! I'll re-open this to remind myself to do it at some point!

closed 2018-01-25 by Gadgetoid