pinout-xyz/Pinout.xyz #138: How do I mark a pin as Input OR Output

4tronix, 2016-10-23:
Some pins are neither 100% Inputs, nor 100% outputs eg:
- Ultrasonic sensor on Picon Zero is an Output and an Input for Sending or receiving pins
- Some pins are brought from the GPIO connector onto the board "as is" so can be used how you like, but they are not dedicated in anyway
So how can we define those?

RogueM, 2016-10-23:
1) the pin 'direction' can be specified when enumating the it, after it's name, like so:

`22':
    name: LCD CMD/DATA
    mode: output
    active: high`

... it is currently data that is unused and will not show up on the current site. It may be used in the future but is not critical to specify. Or you could specify it as part of the name field if you really want to display it (not recommended).

2) breakout pins should never be specified.

they play no part in the functionality of the add-on, and would clutter the pinout as well as wrongly imply wrongly that some add-ons are not possible to combine with others (or rather that could be a problem for the least experienced users).

4tronix, 2016-10-23:
1) Pin direction "is not used", so to specify that it can be either or both of Output or Input then I can use "Input/Output" without causing a problem, right?
2) Agreed

RogueM, 2016-10-23:
1) yes... I'm trying now to be sure it won't trip the html renderer but it should be fine

RogueM, 2016-10-23:
yes, it works:

```python
  '22':
    name: 'birectional pin'
    mode: input/output
```

4tronix, 2016-10-25:
> yes, it works:
>
>   '22':
>     name: 'birectional pin'
>     mode: input/output

Great. Thanks.

closed 2016-11-03 by RogueM