pinout-xyz/Pinout.xyz #139: Would be Niceot Compare/Overlay Multiple Boards

4tronix, 2016-10-23:
I'm often ask if board xx will work with board yy. In fact, this seems to be a common use for pinout.xyz
Wouldn't it be nice if you could overlay two (or more?) boards and see where there were clashes?

RogueM, 2016-10-23:
that's a long term plan of the site, yes. @Gadgetoid even just put a loose timeline on it: https://github.com/Gadgetoid/Pinout.xyz/projects/2

lurch, 2016-10-23:
It's also mentioned in https://github.com/Gadgetoid/Pinout.xyz/blob/master/README.md#roadmap--wishlist  :-)

Gadgetoid, 2016-10-23:
I'm glad to see more support in favour of this idea. It's been on my TODO list since I started adding boards to Pinout. The main reason it hasn't been done already is that Pinout is statically generated, and I haven't yet come up with a way I'm satisfied with to provide this functionality.

The two options are:

1. Static generate json files to define the pinout of each board, and create a comparison tool in Javascript that runs in the users browser
2. Store the board details in a database and create a server-side comparison tool

I'm more in favour of option 1, but it has its drawbacks and relying on Javascript is not ideal in the geekiverse :D At the moment that's what the boards page does, and it's already clunky.

lurch, 2016-10-23:
I guess it depends whether you want the 'cpu load' to be on the users' machines, or your webserver ;-)

Gadgetoid, 2016-10-23:
Ideally, their machines ;) I think the server would just be crushed under the weight of anything dynamic, it's a can of worms I'd rather not open :D

closed 2017-11-29 by Gadgetoid

lurch, 2017-11-30:
Does this issue being closed mean that it's never going to get implemented? :-(

lurch, 2017-11-30:
Oh, I see this is now a duplicate of #235 
Sweet!

Gadgetoid, 2017-11-30:
Aye aye! I absolutely plan for a proper, non-branded compatibility tool to be a thing, but for the time being it was easier to do it in collab with Pimoroni since we have a good idea of what works with what, and good data for our boards. And pHAT Stack was a super good excuse!