pinout-xyz/Pinout.xyz #141: 'Ground' highlight colour

lurch, 2016-10-23:
While looking at http://pinout.xyz/pinout/dots I noticed that when you mouse-hover over a 'Ground' pin, the background-colour for the pin changes to the same background colour as all the other (non-selected) pins. (yes, I realise the 'blob' changes colour as well, but I'm just talking about the background behind the text-label for now)

Perhaps it would be more consistent for the mouse-over-background-colour for ground (for the ground pins that are actually used) to be the same slightly lighter shade that the other non-ground pins (that are actually used) use for their mouse-over-background-colour? :)

Gadgetoid, 2016-10-23:
Agreed. Looking over that page, it looks like the second 3v3 power pin has gone wrong too, and they suffer from the same problem.

lurch, 2016-10-23:
Well according to the 'blobs' pin 17 isn't used on the DOTS-hat, so it's probably correct in showing the 'normal' shade on mouse-hover?
This is what it looks like when pin17 *is* used: http://pinout.xyz/pinout/automation_hat

Gadgetoid, 2016-10-27:
I believe I've fixed the styling niggles in my latest bout of CSS hackery. I've cut out some dead CSS, hacked back lots of over-qualified selectors and generally made it less of an awful mess. It's still a mess, but now it's less of one :D 

The majority of tweaking was in 609aad6dc9d0dc0c09a5d895726c372ee5034de4

lurch, 2016-10-28:
Hmmm, previously when a pin was unused by a board, it showed the 'darker' beige when mouse-hovering over it, with the 'lighter' beige being used _only_ for pins used by that board. But now it seems like **all** pins have the lighter beige on mouse-hover, regardless of whether they're used by the board or not?

> It's still a mess, but now it's less of one :D

I can't judge, because I'm only a Python-monkey and Perl-monkey, not a CSS-monkey too :)

Gadgetoid, 2016-10-28:
Oh I saw the darker beige as a bug and fixed it. It was way too subtle to ever have been intentional. Still, this doesn't preclude me from agreeing with you in general. I'll see what I can come up with!

Gadgetoid, 2016-11-02:
I'm not quite happy with the colours as they are, but I'm also not in the right mindset to fix them. I'll tuck this issue away for now, but sooner or later I hope I'll have an "aha" moment. Either way I don't think it's a major problem, but I think the look and feel of the pinout is starting to irritate me.

Things that bug me:

* Physical numbers are necessary, but are quite awkwardly crammed in
* Half the pinout is on a green background, the other half on a white background
* Board views look quite busy, with many highlighted pins
* Other

closed 2016-11-02 by Gadgetoid

lurch, 2016-11-02:
I've just noticed that on e.g. http://pinout.xyz/pinout/servo_pwm_pi_zero the 'used' power pins no longer highlight when you mouse-over them, which seems a bit inconsistent.

> Physical numbers are necessary, but are quite awkwardly crammed in

Maybe add a couple more pixels of width between the 'header' and the edge of the board (IYSWIM) so that the two-digit even-pin physical numbers look less squished?

> Half the pinout is on a green background, the other half on a white background

IMHO it's a useful indicator of which way around the board is though.

> Board views look quite busy, with many highlighted pins

Looks fine to me. Perhaps I've just got "too used to it"?

I've just noticed that if you click the forward and back buttons in your browser, to switch back and forth between http://pinout.xyz/pinout/servo_pwm_pi_zero and http://pinout.xyz/ , then the pinouts-section of the page seems to jump left and right a bit, which looks strange.
Although bizarrely, the same doesn't happen when switching between http://pinout.xyz/pinout/wifi_pants and http://pinout.xyz/ !