pinout-xyz/Pinout.xyz #150: Getting translations up-to-date

RogueM, 2016-11-03:
>>> Our German-language site is by far the most popular localisation. I'd like to a) See it updated, and b) Try to better disseminate the localised versions

trying out the converting notes to issue.

RogueM, 2016-11-03:
@Gadgetoid I suggest to tap the original translators on the shoulder and see if they could give it a fair push over the next few weeks / over Christmas?

Gadgetoid, 2016-11-26:
That's not a bad idea. The Christmas period should be a good time to poke people :D I think there are some more boards inbound too, but that doesn't matter so much now we've got your system for managing them.

RogueM, 2016-11-26:
yes, though do remember that new pushes online should be for all LANG, otherwise the system does not quite work ;-)

... well, at least in term of enticing new translators lending a hand.

RogueM, 2016-11-26:
this also leads to 404:
http://de.pinout.xyz/pinout/pan_tilt_hat

Gadgetoid, 2017-01-02:
D'oh! Despite lagging behind, Germany is still our largest secondary language. But they all need some TLC. I had a bit of a lazy December though =/ very, very unproductive. (I needed it, I think)

Spanish language is not as big as it should be- I suspect because the Pi hasn't reached into Spanish-speaking countries as well as it could have.

diegorodriguezv, 2017-02-15:
Hi.

I'll try to translate a few of the boards at  src/es/translate.

Is there any guide somewhere for the process? Or it's simply a pull request?

RogueM, 2017-02-15:
the files needing translation can be found in the `translate` dir for your LANG:
https://github.com/Gadgetoid/Pinout.xyz/tree/master/src/es/translate

... modify them in place + PR yes (we'll take care of merging them in)

for typos/error in already translated pages, modify in place + PR:

https://github.com/Gadgetoid/Pinout.xyz/tree/master/src/es/overlays

closed 2018-01-25 by Gadgetoid