pinout-xyz/Pinout.xyz #151: Pin information for the PiTFT

BruceJohnJennerLawso, 2016-11-04:
As far as I know the PiTFT is probably the most well known display board besides the official screen, it would be nice to have it added to the list of boards.

Some useful information about the pin usage:

https://forums.adafruit.com/viewtopic.php?f=47&t=75974

geek2003, 2016-11-06:
here is as much as I have managed to do can anyone add it to the site?

[image]
[tft.txt](https://github.com/Gadgetoid/Pinout.xyz/files/573857/tft.txt)

geek2003, 2016-11-06:
the main problem is there are so many versions of the pi TFT

Gadgetoid, 2016-11-14:
If we name them followed by size and resolution it may help. Or we can simply be opinionated and list just one or two. I think most/all of them use the same pins for the display signal, though?

RogueM, 2016-11-14:
I'm not sure if there are differences to the pinout... would need checking. There is no need for a custom kernel at this point though, so install instructions should be toned down to a sane level before pushing live ;-)

RogueM, 2016-11-14:
the PiTFT+ 3.5 will be in the next update (commit https://github.com/Gadgetoid/Pinout.xyz/commit/ad33003efe65585be5d977428b5072011ecbc66c )

... will have to check whether others have the same pinout and could be merged into it, or ned their own overlay.

closed 2016-11-14 by RogueM

BruceJohnJennerLawso, 2016-11-16:
Apologies for slow reply on this.

If we include nothing else, the 2.8 resistive & capacitive boards from adafruit as well as the 3.5 would be good to include. There are a multitude of knockoff tfts out there, but the adafruit ones are far away the most common.