pinout-xyz/Pinout.xyz #152: Ground page

geek2003, 2016-11-05:
Just a little thing but on pinout.xyz/pinout/ground the ground pins along the left hand side are different and you can't see the text of ground. As I said it is a little thing but worth fixing!

lurch, 2016-11-06:
I guess this must a side-effect of the changes made for #141

Gadgetoid, 2016-11-08:
Yup, I believe I've broken this in my efforts to reign in the horrible nightmare hole of CSS I'd dug myself into. Should be an easy fix! Thanks for pointing it out.

lurch, 2016-11-08:
Just spotted another small "problem", which is nicely illustrated by http://pinout.xyz/pinout/blinkt
The 'unused' Ground pins still have the black circle drawn around them (good), but the used ground pins don't have any circle around them, which makes them look weird.

Gadgetoid, 2016-11-26:
I think this has also been fixed somewhere along the way, thanks. I have to squint to see the black circles :D

closed 2016-11-26 by Gadgetoid