pinout-xyz/Pinout.xyz #153: Graphical pinout missing SPI1 CE-pins

WereCatf, 2016-11-05:
The graphical pinout is real nice, but, as per the title, while it does show MISO, MOSI and SCLK for SPI1, it doesn't show the CE-pins for it. Minor oversight, but perhaps worth fixing.

Gadgetoid, 2016-11-08:
The main view tries to capture some of the more common pin usages, so favours PWM over SPI1 CE0. For everything else- it's designed to be interactive, so you can usually find the information you need in a click or two. 

For example, you can still find full details of SPI pins here: http://pinout.xyz/pinout/spi

RogueM, 2016-11-08:
still, BCM16 could be highlighted as a CE pin for SPI1?

Gadgetoid, 2016-11-26:
True! I'll flag this up to do ;D

Gadgetoid, 2019-01-26:
I never did get 'round to this, and looking back over it the fact SPI1 CE0 conflicts with the pin currently labelled as "PWM" makes me hesitant to bother with the SPI1 CE pins at all. Closing for the time being!

closed 2019-01-26 by Gadgetoid