pinout-xyz/Pinout.xyz #160: wirinpi BCM0 and BCM1

robotrovsky, 2016-11-24:
Hi!

First i want to say, that i am using pinout.xyz very frequently and that it is a very nice tool!

In the wiringPi-overlay, there are missing the wiringPi Pins for BCM 0 and BCM 1. These are:

BCM0 = WriingPi Pin 30
BCM1 = WriingPi Pin 31

Please look at the translation in wiringPi.c Line 354:
https://github.com/WiringPi/WiringPi/blob/master/wiringPi/wiringPi.c#L354

Gadgetoid, 2016-11-26:
You're not wrong. These are "officially" don't-connect pins, but I don't see any harm in adding the WiringPi names for them.

Thanks for your feedback and support!

RogueM, 2016-11-28:
interestingly those were documented under the physical pins. anyhow, added (will go live whenever next sync occurs).

closed 2016-11-28 by RogueM

robotrovsky, 2016-11-29:
Thanks to you guys, for the nice tool!