pinout-xyz/Pinout.xyz #164: Adafruit ultimate GPS hat complete file

geek2003, 2016-12-04:
here it is. I promised more adafruit boards.

[gps.txt](https://github.com/Gadgetoid/Pinout.xyz/files/629755/gps.txt)
[image]

not really an issue just don't know how to do pr's!

More on the way!

geek2003, 2016-12-04:
[image]
 sorry this is a better image

geek2003, 2016-12-08:
Hello?

RogueM, 2016-12-08:
Yes, thanks, noted... we'll review and publish it ASAP.

geek2003, 2016-12-08:
Cheers! Looking to do one for the new adafruit audio phat/bonnet so expect it soon!

RogueM, 2016-12-30:
done and ready to go.

closed 2016-12-30 by RogueM