pinout-xyz/Pinout.xyz #167: Missing: "Raspberry Pi High-Precision AD/DA Expansion Board"

emmellsoft, 2017-01-03:
Hello pinout-people!   :-)
I didn't find this board on your https://pinout.xyz/boards page:

http://www.waveshare.com/High-Precision-AD-DA-Board.htm

RogueM, 2017-01-08:
thanks for the request... there is enough info there so it should be possible to add.

If you are able to submit a draft overlay via PR though this will ensure it happens more rapidly as we don't often have time to assemble the needed info*.

* the average it takes me to make a board from scratch is about 30 mins, but someone more familiar with it may take significantly less without compromising accuracy - either way, time is the bottleneck at our end.

RogueM, 2017-01-08:
actually the relevant link is this:
http://www.waveshare.com/wiki/High-Precision_AD/DA_Board

Gadgetoid, 2018-04-08:
https://pinout.xyz/pinout/high_precision_adda_board

closed 2018-04-08 by Gadgetoid