pinout-xyz/Pinout.xyz #173: Graphical pinout missing SDIO & I2S busses.

campag, 2017-01-12:
It would be helpful to add those two pin mappings, as it would help (p)HAT designers avoid trashing those busses by using their pins as arbitrary GPIO. I'm looking at you, you naughty pirates.

There won't be many pins left untagged, but those are the ones to use (perhaps jumper selectable even) for maximum co-operability with other boards.

I2S overlaps SPI1 - choice to be made there.

Gadgetoid, 2017-02-15:
:P 

Many of the products that use those pins use almost *every* pin!

It's a bit of a tricky one, since they share some functions. You can use:

* http://pinout.xyz/pinout/pcm
* http://pinout.xyz/pinout/spi

To see the protocols in isolation. But I appreciate why you'd want to see it at-a-glance.

I think all of the PCM/i2s pins are already tagged up as a mixture of SPI and PWM too.

campag, 2017-02-15:
Definately PCM over 2nd SPI - you can bit-bang SPI on arbitrary GPIO if one i/f isn't enough.

Something like this:
[image]

SDIO can use 4 or 6 pins in 1-bit or full 4-bit modes, so I showed D2 & D3 differently - you can use 26/27 as arbitrary GPIO while still having useful quarter-speed SDIO.

I think that makes it a bit clearer which pins you can use safely without punching a hole in some other peripheral bus.

lurch, 2020-08-30:
Maybe there's a slight danger of "information overload" in the diagram? Given that each Pin has several different ALT functions there's always bound to be conflicts, and I suspect that for the **vast** majority of pinout.xyz users, there's no need to know which pins to SDIO bus is available on?
This issue is also kinda similar to #372

axelsimon, 2020-12-03:
One option in this case is two have two versions of the diagram, a normal (simplified) one, which is the one showed on the page by defaulte, and a full-on version, with all the details, shown only when the mouse cursor is over it.

Gadgetoid, 2026-08-18:
Closing as stale, the image is very much a secondary concern to the website proper and the source files are (IIRC) available and the license permissible for anyone who wants to mangle it to their own nefarious purposes. (IMO its an awful pinout diagram that serves no real purpose but 🤷 )

closed 2026-08-18 by Gadgetoid