pinout-xyz/Pinout.xyz #175: Add Sheepwalk Electronics Boards

njh, 2017-01-13:
Hello,

I am a Sheepwalk Electronics customer/fan:

They make three 1-wire boards for Raspberry PI:
* RPI1 GPIO Host Adapter for Raspberry Pi
* RPI2 1-Wire Host Adapter for Raspberry Pi (uses i2c -> 1wire)
* RPI3 1-Wire Host Adapter for Raspberry Pi Models A+, B+, 2, 3 (multi-bus)

http://www.sheepwalkelectronics.co.uk/index.php?cPath=22&osCsid=2755a5605f1cef8c2a682d7aefb73d6f

RPI2 board was submitted in PR #170 and then moved to Draft.

I don't have top-down photos of boards RPI1 and RPI3.

nick.

njh, 2017-02-09:
Any news on this?

Is there anything I can do to help?

Gadgetoid, 2017-02-15:
Not yet, sorry, I've been a bit lax pushing out updates recently but I'm keeping my eye on things. A mental note has been duly made!

Gadgetoid, 2017-08-01:
At this point someone needs to reach out to the manufacturer and ask for good quality top-down photos of the boards. Either that, or an existing owner could snap some, or they could send us some boards to photograph here.

njh, 2017-08-01:
Ok, if image quality is the issue, I will see what I can do.

njh, 2017-08-09:
Hi,

I took a better photo:
[image]

Is it good enough?

Originals are in this Dropbox folder:
https://www.dropbox.com/sh/rj243vhx9i24x4c/AAB8gKngES1bcx0pRPJ2M1Vca?dl=0

nick.

Gadgetoid, 2026-08-18:
Closing as stale. I have no bandwidth to manage board additions, but I do try to merge contribs when I can!

closed 2026-08-18 by Gadgetoid